commaai/openpilot #34879: Hyundai controlsMismatch on nightly

adeebshihadeh, 2025-03-15:
```
| 16 | 0ae0c64127c77e40/00000000--7bee84686d |        0 |         0 | HYUNDAI_KONA_EV_2022 | 5c30208      | 0.9.9-release | threex        | event controlsMismatch |
```

sshane, 2025-03-18:
Running this segment offline, safety says there's no invalid rx addresses. I even ran on the route's source commit

Jon8RFC, 2025-04-22:
c1fc13c1b806e536
Comma3 dongle ID

Not sure if this is the same thing, and the logs are now gone but next drive I'll remember mark them to be preserved, hopefully.

My 2018 Hyundai Sonata (SONATA_LF) gets frequent AEB/LKAS since updating to 0.9.8.

https://youtube.com/shorts/e_p5m_zWgZk

sshane, 2025-05-28:
@Jon8RFC can you try master or nightly? We fixed an issue that caused LKAS faults on startup when your device was off, such as the first drive of the day with a comma power.

Have you hit any issues since?

closed 2025-05-28 by sshane

Jon8RFC, 2025-05-28:
@sshane I can't use my car for the next day or two, or maybe longer, unfortunately.  Thanks for letting me know.

Jon8RFC, 2025-06-02:
@sshane I haven't had issues since cleaning the USB port and cable with contact cleaner, before trying the nightly.
I did run the nightly recently though, and I've uploaded the logs if you want to grab those before they disappear.

Dongle ID  c1fc13c1b806e536