commaai/openpilot #34916: Camera Malfuntion since 0.9.8 update. Both road camera and wide camera

sweetrap1, 2025-03-21:
### Describe the bug

Since the 0.9.8 update I get a Camera Malfunction error. Openpilot is not operatable. Here is the segment ID  Dongle ID: 394ac061420e5cdc

### Provide a route where the issue occurs

394ac061420e5cdc/00000004--1de6a48d26

### openpilot version

0.9.8

### Additional info

I have flashed the device with the new firmware, tried the new sunnypilots etc. If I install an older version of 0.9.7 of open pilot it works just fine with no errors. If I try to engage the system while it is saying camera malfunction, I get another error, either cruise fault, restart vehicle (that doesn't work) or radar fault.

sweetrap1, 2025-03-21:
I re-flashed the device, unplugged the power for 2 hours and reinstalled openpilot. I still get the same Camera Malfunction error. Here is the segment ID. 394ac061420e5cdc/00000000--07e114da6f/0

adeebshihadeh, 2025-03-21:
Thanks for trying that. I'll look into this soon, but in the meantime, I suggest reverting back to 0.9.7 since it may take a while to look into since only one other user is hitting this.

sweetrap1, 2025-03-22:
Sounds like a plan. I will revert back. Let me know if you want to me try anything else, or want more data. I'm more than willing to help.

sunnyhaibin, 2025-03-23:
Another user is also hitting this with an official C3X: https://github.com/sunnypilot/sunnypilot/issues/703#issuecomment-2746242645

> C3X info:
The Dongle ID : 647c045f0ff7f1fe
The serial : 1a42809d

geniuth2, 2025-03-25:
I have a question about camera malfunction.
My com3x has camera problem. In addition, my com3x has same problem with 0.9.7. It may hardware problem.
I would like to send my com3x to COMMA to fix.

incognitojam, 2025-03-25:
@geniuth2 Please check out https://comma.ai/support and follow all the troubleshooting steps then you can open a ticket

SG5, 2025-03-27:
I can confirm same problem. I have 0.9.8 for about a week. This morning got both cameras don't show anything

adeebshihadeh, 2025-04-01:
@sweetrap1 can you give `installer.comma.ai/commaai/nowait` a try?

sweetrap1, 2025-04-01:
@adeebshihadeh That worked and fixed the camera issues for me. Overall, it drove nicely. Besides, halfway into my test drive, I got a Take Over Immediately warning for a slow communication between processes error. https://connect.comma.ai/394ac061420e5cdc/00000000--21253c823a

adeebshihadeh, 2025-04-01:
Fixed by #34972. It should go out to the `nightly` branch within a day, then it'll go to release pretty soon after. I'll leave this open until it gets to release.

adeebshihadeh, 2025-04-07:
Fix pushed to release!