commaai/openpilot #34922: Limit excessive transfer for those with data caps

bhajneet, 2025-03-22:
The comma ai is using around half a terabyte of data each month. This is really not good for those that have to pay overage fees for their internet. Please allow some kind of limit or compression to lower this value, whatever it is.

[image]

sshane, 2025-05-16:
@deanlee interested in this one? We want a toggle that overrides Network Manager's metered setting per network

deanlee, 2025-05-17:
@sshane  I’m concerned that having a metered setting for each individual WiFi network might introduce unnecessary complexity.
How about adding a single toggle in the advanced network settings to enable or disable the metered status for all WiFi connections? This approach would be simpler and still effective, without needing to manage settings for each specific connection. What do you think?

adeebshihadeh, 2025-05-17:
Oh, NetworkManager should handle keeping the state.

deanlee, 2025-05-17:
It looks like uploads on metered connections are only restricted for  `/crash ` `/boot ` and `qcamera.ts`, other log files seem unrestricted. should we limit uploads for these files to save bandwidth?

```python
       # limit uploading on metered connections
        if metered:
          dt = datetime.timedelta(hours=12)
          if logdir in self.immediate_folders and (datetime.datetime.now() - datetime.datetime.fromtimestamp(ctime)) < dt:
            continue

          if name == "qcamera.ts" and not any(logdir.startswith(r.split('|')[-1]) for r in requested_routes):
            continue

        yield name, key, fn
```

adeebshihadeh, 2025-05-17:
No, that's by design.

sshane, 2025-05-19:
We need to affect `HARDWARE.get_network_metered(network_type)`, you don't have to think about uploading

sshane, 2025-05-23:
https://github.com/commaai/openpilot/pull/35293

closed 2025-05-23 by sshane