commaai/openpilot #34923: Docker Build Fails with 'QGeoCoordinate' Not Found Error for openpilot v0.9.7

jeffxhsiao, 2025-03-22:
### Describe the bug

I downloaded openpilot v0.9.7 and attempted to install it in a Docker environment using the command:
`docker build -t openpilot:0.9.7 -f Dockerfile.openpilot .`

However, during the Docker build step `RUN scons --cache-readonly -j$(nproc)`, I consistently encounter the following error:
```
In file included from selfdrive/navd/moc_map_renderer.cc:1:
./selfdrive/navd/map_renderer.h:9:10: fatal error: 'QGeoCoordinate' file not found
9 | #include <QGeoCoordinate>
   |          ^~~~~~~~~~~~~~~~
```
[image]
[image]
Could someone please advise on how to resolve this issue or suggest the correct way to install openpilot v0.9.7 within Docker?
OS Version: Ubuntu 22.04

### OS Version

Ubuntu 22.04

### openpilot version or commit

openpilot v0.9.7

jeffxhsiao, 2025-03-23:
I have already resolved the above 'QGeoCoordinate' Not Found Error, but now I've encountered a new Dockerfile installation error:

```
8.892 Traceback (most recent call last):
8.892   File "/home/batman/openpilot/selfdrive/controls/lib/longitudinal_mpc_lib/long_mpc.py", line 11, in <module>
8.893     from openpilot.selfdrive.car.interfaces import ACCEL_MIN
8.893   File "/home/batman/openpilot/openpilot/selfdrive/car/interfaces.py", line 20, in <module>
8.893     from openpilot.selfdrive.controls.lib.events import Events
8.893   File "/home/batman/openpilot/openpilot/selfdrive/controls/lib/events.py", line 9, in <module>
8.893     import cereal.messaging as messaging
8.893   File "/home/batman/openpilot/cereal/messaging/__init__.py", line 2, in <module>
8.893     from msgq.ipc_pyx import Context, Poller, SubSocket, PubSocket, SocketEventHandle, toggle_fake_events, \
8.893   File "/home/batman/openpilot/msgq/__init__.py", line 2, in <module>
8.893     from msgq.ipc_pyx import Context, Poller, SubSocket, PubSocket, SocketEventHandle, toggle_fake_events, \
8.893 ModuleNotFoundError: No module named 'msgq.ipc_pyx'
8.929 scons: *** [selfdrive/controls/lib/longitudinal_mpc_lib/c_generated_code/Makefile] Error 1
```

[image]

adeebshihadeh, 2025-06-14:
Closing as this isn't a bug - check out our [Discord](https://discord.comma.ai) instead.

closed 2025-06-14 by adeebshihadeh