commaai/openpilot #34924: Add support for custom eSIM profile loading

adeebshihadeh, 2025-03-22:
comma 3X ships with an eSIM used for the comma prime service, however it's also capable of loading up an profile like your phone. This bounty is for making it possible for users to load their own SIM profiles.

This is a multi-parter:
1. add profile loading capability to openpilot
2. expose it over athena
3.  make a good experience in [connect](https://github.com/commaai/connect)

See https://github.com/commaai/openpilot/issues/32717 for context. Must use https://github.com/estkme-group/lpac for the provisioning (i.e. don't use `system/hardware/tici/esim.py` since we can't rely on the EG25's built-in LPA).

greatgitsby, 2025-05-01:
[eSIM profiles for testing](https://source.android.com/docs/core/connect/esim-test-profiles)

greatgitsby, 2025-05-03:
# testing with lpac

grab a recent `lpac` build, like: https://github.com/estkme-group/lpac/actions/runs/14394325163

run on device:
```
#!/bin/bash

#export LIBEUICC_DEBUG_APDU=1
export LPAC_APDU=at
export AT_DEVICE=/dev/ttyUSB2

systemctl stop ModemManager
./lpac $@
```

greatgitsby, 2025-05-03:
main issue right now is that the profile enable does not persist. profile enable, then restarting ModemManager, `system/hardware/tici/restart_modem.sh`, or rebooting system all cause the profile to be reset to the original one that comes with the 3x.

welcome to any other input / attempts at resolving this! i may not be understanding what the normal SOP should be.

adeebshihadeh, 2025-05-03:
Does `esim.py` behave the same way? I don't recall that happening when I wrote that script.

greatgitsby, 2025-05-03:
modified slightly to "enable" instead of "download" first arg and this is what I get:

```
comma@comma-bb546e42:/data/openpilot$ RESTART=1 system/hardware/tici/esim.py ---
Stopping LTE...
 Waiting until the modem comes online...
  Modem is online!
 Pulsing power button...
 Waiting until the modem goes offline...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Modem is offline!
Starting LTE...
 Resetting...
 Pulsing power button...
 Waiting until the modem comes online...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Waiting...
  Modem is online!
==> AT
<== 'AT\r\r\nOK'
==> AT+QESIM="list"
<== 'AT+QESIM="list"\r\r\nERROR'
['', 'ERROR']
==> AT+QESIM="enable","---"
<== 'AT+QESIM="enable","---"\r\r\nERROR'
==> AT+QESIM="list"
<== 'AT+QESIM="list"\r\r\nERROR'
['', 'ERROR']
```

greatgitsby, 2025-05-03:
spoke offline, the current Quectel firmware does not work with `esim.py`.

see discord for download: https://discord.com/channels/469524606043160576/1368111766784966696/1368363312043004156

greatgitsby, 2025-05-19:
profile download example:
```bash
comma@comma-bb546e42:/data/openpilot$ sudo LPAC_APDU=qmi QMI_DEVICE=/dev/cdc-wdm0 lpac profile download -a 'LPA:1$rsp.truphone.com$QRF-BETTERROAMING-PMRDGIR2EARDEIT5'
{"type":"progress","payload":{"code":0,"message":"es10b_get_euicc_challenge_and_info","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es9p_initiate_authentication","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es10b_authenticate_server","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es9p_authenticate_client","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es8p_meatadata_parse","data":{"iccid":"8944476500001944011","serviceProviderName":"BetterRoaming","profileName":"BetterRoaming","iconType":null,"icon":null,"profileClass":null}}}
{"type":"progress","payload":{"code":0,"message":"es10b_prepare_download","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es9p_get_bound_profile_package","data":"rsp.truphone.com"}}
{"type":"progress","payload":{"code":0,"message":"es10b_load_bound_profile_package","data":"rsp.truphone.com"}}
{"type":"lpa","payload":{"code":0,"message":"success","data":null}}
```

greatgitsby, 2025-05-19:
nickname:
```bash
comma@comma-bb546e42:/data/openpilot$ sudo LPAC_APDU=qmi QMI_DEVICE=/dev/cdc-wdm0 lpac profile nickname XXXXXXXXXXXXXXXXXXX 'trey airalo'
{"type":"lpa","payload":{"code":0,"message":"success","data":null}}
```

greatgitsby, 2025-11-25:
after more thorough testing, i haven't been able to get user-added profiles enabled alongside the manufacturer provisioned profile on the eUICC.

third-party profiles will work on device (i've tested this extensively on my three x). however, users must wipe the eUICC's memory using an SGP.22 compliant LPA (Local Profile Assistant) before they can enable any other stored profile and have it remain active without reverting to the comma profile. see the [`bootstrap` process in `esim.py`](https://github.com/commaai/openpilot/blob/f01391a7d9b5d0b7a56a346d85c7de42302656b7/system/hardware/tici/esim.py#L67-L72) on how this works successfully today. while this is somewhat tenable on the 3X with its easily removable SIM card, it is not on the comma four with the soldered on eUICC. it's certainly not user friendly in either case; if a customer wants to switch back to full comma prime, then they need to purchase another SIM card from comma.

i believe the root cause is tied to the chip's profile being flashed with an SGP.32 *IPA* (IoT Profile Assistant, *not* LPA) which includes extra specs like fallback mechanisms and emergency profiles designed for IoT devices in the field via remote management. these processes aren't defined in the SGP.22 spec and are not modifiable locally because the modification of these fallback mechanisms are delivered via packages signed by the SM-DP+ server, which are cryptographically verified by the eUICC chip itself.

this might be worth investigating further internally, but outside contributors will likely get stuck where I did since the manufacturer provisioned profile and management are a black box to us.

open questions:
- can you disable the fallback / emergency profile on the management platform?
- is there a way for customers to wipe the eUICC anyway and let comma re-provision the eUICC chip? (I imagine a beautiful one-click recovery in comma connect)

---

below is the reproduction of this fallback behavior in the field using [my SGP.22 compliant LPA impl](https://github.com/greatgitsby/openpilot/blob/qrvibes/system/hardware/tici/lpa.py) (but same behavior is seen with other LPAs like `lpac`). note that `ModemManager` was masked at this time and not running.

```
comma@comma-b77675ea:/data/openpilot$ date; python system/hardware/tici/lpa.py --enable 8948010000069391043
Tue Nov 25 07:32:08 PM UTC 2025
profile 8948010000069391043 already enabled
[
  {
    "iccid": "89852351224030088594",
    "isdpAid": "A0000005591010FFFFFFFF8900001001",
    "profileState": "disabled",
    "profileNickname": null,
    "serviceProviderName": "Webbing",
    "profileName": "Webbing",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  },
  {
    "iccid": "8948010000069391043",
    "isdpAid": "A0000005591010FFFFFFFF8900001200",
    "profileState": "enabled",
    "profileNickname": null,
    "serviceProviderName": "Sparks",
    "profileName": "Sparks",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  }
]

comma@comma-b77675ea:/data/openpilot$ date; python system/hardware/tici/lpa.py --process-notifications
Tue Nov 25 07:32:28 PM UTC 2025
Found 2 notification(s) to process
Processing notification seqNumber=17, address=rsp.truphone.com
Notification 17 processed successfully
Processing notification seqNumber=18, address=rsp.truphone.com
Notification 18 processed successfully

^[[Acomma@comma-b77675ea:/data/openpilot$ date; python system/hardware/tici/lpa.py
Tue Nov 25 07:32:32 PM UTC 2025
[
  {
    "iccid": "89852351224030088594",
    "isdpAid": "A0000005591010FFFFFFFF8900001001",
    "profileState": "disabled",
    "profileNickname": null,
    "serviceProviderName": "Webbing",
    "profileName": "Webbing",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  },
  {
    "iccid": "8948010000069391043",
    "isdpAid": "A0000005591010FFFFFFFF8900001200",
    "profileState": "enabled",
    "profileNickname": null,
    "serviceProviderName": "Sparks",
    "profileName": "Sparks",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  }
]

comma@comma-b77675ea:/data/openpilot$ date; python system/hardware/tici/lpa.py
Tue Nov 25 07:32:38 PM UTC 2025
[
  {
    "iccid": "89852351224030088594",
    "isdpAid": "A0000005591010FFFFFFFF8900001001",
    "profileState": "enabled",
    "profileNickname": null,
    "serviceProviderName": "Webbing",
    "profileName": "Webbing",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  },
  {
    "iccid": "8948010000069391043",
    "isdpAid": "A0000005591010FFFFFFFF8900001200",
    "profileState": "disabled",
    "profileNickname": null,
    "serviceProviderName": "Sparks",
    "profileName": "Sparks",
    "iconType": null,
    "icon": null,
    "profileClass": "operational"
  }
]
```

greatgitsby, 2025-12-03:
lil bit of magic on comma end, and it works now:

[image]

for posterity, SIM has to be put into "manual" mode and the auto-fallback feature is disabled for the eUICC.

greatgitsby, 2026-09-09:
complete with https://github.com/commaai/openpilot/pull/38813

greatgitsby, 2026-09-09:
openpilot now has both esim.py CLI and comma four UI to install, switch, and manage profiles.

@adeebshihadeh i believe comma 3X UI is in KTLO. i won’t port UI support there. let me know what remaining validation you want on this issue.

valerito, 2026-09-23:
How do I set the eUICC to manual mode?

closed 2026-10-17 by adeebshihadeh

benmarten, 2026-10-26:
> 26\. How do I set the eUICC to manual mode?

I would like to know this too - @adeebshihadeh could you please help? 🙏

benmarten, 2026-11-03:
I actually got it working now - well codex did ;)
I have an sim from montyesim now running and Tailscale which allows me to connect from anywhere.

Codex summary:
```
To enable a custom eSIM and remote SSH on a comma 3X:

Install the eSIM using openpilot’s eSIM CLI/LPA and the carrier’s activation code.
Enable the profile, process pending carrier notifications, configure the carrier’s APN, and enable data roaming.
Handle Webbing auto-fallback. The eUICC can automatically revert to the factory Webbing profile. Ideally, disable fallback through its manual-mode configuration. In our case, deleting Webbing prevented the reversion; we didn’t find a supported local manual-mode command.
Verify cellular connectivity by checking network registration, packet-data attachment, and HTTPS traffic explicitly through ppp0. Restart the modem and confirm automatic reconnection.
Install Tailscale persistently under /data, store its identity there, and add a startup hook. Authenticate it to your tailnet. On this device, Tailscale needed legacy iptables because the default nftables backend was incompatible.
Connect over the Tailscale IP using the device’s existing SSH keys.
Deleting Webbing solved the profile reverting problem. Cellular connectivity still requires a working carrier profile, correct APN, and roaming configuration.

```