commaai/openpilot #34927: camera ui + recording too dark at night

royjr, 2025-03-24:
### Describe the bug

Similar to [#31627](https://github.com/commaai/openpilot/issues/31627), but this time it appears to affect the camera recordings as well — potentially impacting nighttime driving.

- ui vs background

[image]
- recording
[image]

### Provide a route where the issue occurs

e6fd6ef0e9cc1171/0000008e--6b282424ad, e6fd6ef0e9cc1171/000000a9--f6a8e2fbe6

### openpilot version

master

adeebshihadeh, 2025-03-24:
@ZwX1616 can you look into this?

ZwX1616, 2025-03-24:
Scene actually looks fine in the recording with headlights on. This might just be the ui looking too dark when screen is dimmed (screen calib?).
[image]

royjr, 2025-03-24:
But this is the actual scene at midnight, multiple street lamps light up the street.

[image]

closed 2026-02-05 by royjr