commaai/openpilot #34948: Car cutting diagonally across my path

missionkong, 2025-03-28:
Sometimes, when I'm driving, I encounter vehicles cutting diagonally across my path, as shown in the image below. This usually happens when they are trying to cross the road to reach a gas station or a similar destination on the opposite side. I've noticed that OpenPilot does not recognize these vehicles as obstacles and therefore does not slow down. I was wondering if this aspect could be improved.

[image]

sunnyhaibin, 2025-03-30:
Without any logs or segments uploaded, there is nothing to action on or improve.

adeebshihadeh, 2025-03-30:
Thanks for the report, however this should go in the #driving-feedback channel on Discord.

closed 2025-03-30 by adeebshihadeh