commaai/openpilot #34958: comma 3X: auto-configure `adb shell` properly

adeebshihadeh, 2025-03-29:
Go from this mess
```
batman@workstation-adeeb:~$ adb shell
sh-5.2# export TERM=xterm-256color
sh-5.2# su comma
comma@comma-eba4643f:/$ cd /data/openpilot/
comma@comma-eba4643f:/data/openpilot$
```

to
```
batman@workstation-adeeb:~$ adb shell
comma@comma-eba4643f:/data/openpilot$
```

---

More context in https://github.com/commaai/openpilot/pull/34468

JigglyNoahPuff, 2025-04-08:
Would modifying the env to make an alias that does all of this in one command be an acceptable fix?
Also I see that we don't want to assume /data/openpilot exists, what is the desired behavior if that path doesnt exist?

BBBmau, 2025-04-10:
This looks like it could be of use: https://stackoverflow.com/questions/26288222/have-aliases-in-adb-shell-non-interactive-mode

adeebshihadeh, 2025-05-06:
Just added a helper to `adb forward` for a real shell: https://github.com/commaai/openpilot/pull/35135/.

I'd still like to get this fixed up though. I think the way to do this is to hook the shell init and run `su comma && export TERM=xterm-256color`. 

https://github.com/commaai/agnos-builder/blob/master/userspace/files/adbd.service