commaai/openpilot #34969: Hotspot (tethering) mode not working properly after 0.9.8 update

cloudrainstar, 2025-04-01:
### Describe the bug

Steps to reproduce: pop in a SIM card, internet works over lte on comma 3x, then enable tethering, internet stops working on comma 3x, and internet doesn't work on any devices connected to weedle hotspot.
I ssh'd into the comma to check, it seems that in openpilot 0.9.7 the ipv4.route-metric=1100 is correctly shown when checking with the ip route command. However, in openpilot 0.9.8 the metric is set to 600, which is higher priority than the lte route metric at 1000. Doing an ip route del route default via 192.168.43.1 will allow the comma 3x to reach the internet again, but the connected devices will still not have internet.
Additionally, iptables nat rules are set when enabling tethering on 0.9.7, but 0.9.8 iptables have been upgraded to netfilters and I didn't see any nat rules for iptables and iptables-legacy, not sure if that is also part of this issue.

### Provide a route where the issue occurs

N/A

### openpilot version

0.9.8

closed 2025-04-25 by sshane

sshane, 2025-04-26:
The hotspot thus far has been provided as a convenience to ssh into the device, so internet accessibility isn't intended.

Let me know if this fixes it for you! You may need to delete the existing Hotspot connection: `nmcli c del Hotspot`

cloudrainstar, 2025-04-27:
This does indeed fix it to prevent Network Manager creating a default route for Hotspot. Thanks!

For the network sharing (nat) part, it seems network manager can't modify the new iptables-nft due missing nf_tables kernel modules in AGNOS. For a quick workaround using iptables-legacy, I put in a [`sudo iptables-legacy -t nat -A POSTROUTING -s 192.168.43.0/24 -o wwan0 -j MASQUERADE`](https://github.com/cloudrainstar/sunnypilot/commit/dee407b5de4315e9acfd68a9792bc61a250ec734) when tethering is activated, so now my dashcam can send me notifications through the comma 3x hotspot again on 0.9.8+.

Thanks again!