commaai/openpilot #34971: comma 3X: screen driver bug on some newer displays

adeebshihadeh, 2025-04-01:
A handful of recent devices hit a bug in the display driver when the device warms up.

In this state, XBL is always able to bring up the display, but it then hits a bug in the kernel driver later on. We're working on shipping a fix for this as soon as possible.

robbederks, 2025-04-04:
Merged a kernel PR that seems to mitigate this issue on the two test device I have, without having to modify anything else in the boot process: https://github.com/commaai/agnos-kernel-sdm845/pull/85

jyoung8607, 2025-04-05:
> Merged a kernel PR that seems to mitigate this issue on the two test device I have, without having to modify anything else in the boot process: [commaai/agnos-kernel-sdm845#85](https://github.com/commaai/agnos-kernel-sdm845/pull/85)

I think 69dbf99a5c788346 was affected, I could see a display driver panic in the ramoops buffer. Only a little testing time so far, but it's definitely doing better with the AGNOS update.

closed 2025-04-06 by adeebshihadeh

adeebshihadeh, 2025-04-07:
Fix pushed to release!