commaai/openpilot #35010: Shorter logs missing in connect.comma,ai

mvl-boston, 2025-04-11:
### Describe the bug

A number of my shorter logs are missing in connect.comma.ai.   Usually they represent immediate car faults as I attempt to port new cars.

The lack of accessibility in connect.comma.ai is blocking my ability to upload rlogs to debug car port issues.

These logs are available in useradmin (but I can't request upload of rlogs from that site).

### Provide a route where the issue occurs

ad9840558640c31d/0000004d--daf90bf8ea

### openpilot version

0.9.9

incognitojam, 2025-04-12:
In useradmin, there is a link on each route page to open the same route in connect, then you should be able to upload logs that way.

sshane, 2025-04-12:
This is because the time is missing in these routes:

[image]

They will now show up in new connect, but we need a way to handle the bad time. @incognitojam fall back to upload time? Or if end time is there, extrapolate back