commaai/openpilot #35039: Standard openpilot PC setup

adeebshihadeh, 2025-04-20:
$999 is still a bit much for casual interest in openpilot. Let's develop a cheap kit to get driving on openpilot master, with a nice upgrade path to a full comma 3X when you're ready.

This will be a panda + car harness + webcam plugged into a laptop. A [FrEON](https://medium.com/@kirk_40929/getting-comma-openpilot-running-on-a-used-phone-c72d609cb4d3) for the comma three generation.

The major parts that aren't done yet:
* packaging
  * should all fit nicely with a single wire running down to the laptop
  * fits in a comma 3X mount for easy swapping
* `./launch_openpilot.sh` needs to just work
* good docs

This will be both a product on the comma shop and fully open sourced if you want to DIY. Also open to having these register with our backend for use with connect.

adeebshihadeh, 2025-04-22:
Not sure we'll have time to get around to this for a couple months, so opening this up to the community as a software and hardware bounty.

## Hardware bounty: $1k
Turn this into a real product. The deliverable is a PR to [commaai/hardware](https://github.com/commaai/hardware) that adds the `comma_zero/` folder. At a minimum, this includes:
* full BOM: pick a [webcam](https://github.com/commaai/openpilot/tree/master/tools/webcam), USB hub, etc.
  * the cheaper the better! we want this to be as widely accessible as possible
  * the selected parts should be widely available internationally
* design a beautiful printed housing to tie it all together
  * should have a single USB-C running to the laptop
  * slides into a comma 3X mount
* good docs for how to build it 

[image]

## Software bounty
TODO

---

Once both are done, we'll put it on the shop and write up a nice blog post!

carlin-rj, 2025-04-22:
good job

zailushang2008, 2025-04-22:
good job

adeebshihadeh, 2025-04-25:
HW bounty locked to @lukasloetkolben! Prototype is being discussed in Discord

J-Palomino, 2025-04-30:
is anyone familiar with the blockers regarding USB webcam support?

lukasloetkolben, 2025-05-27:
I have already created a PR in the hardware repository. Even if it seems a bit inactive at the moment. I made a video and took pictures to make a guide, so I'm still working on it.
But right now, we have to work out if we can use the camera I chose, because it only records at 30 frames per second.

JacksonRT, 2025-05-27:
@lukasloetkolben Alright

jakethesnake420, 2025-06-18:
Ok so go to Walmart and buy every camera there and see which works the best

carlin-rj, 2025-06-19:
> comma zero 的目的是让你能够以尽可能少的钱（不用花 999 美元以上买 C3X）在你的车上运行 openpilot **master**（主要作为开发者）。而且它的驾驶体验不会像 C3X 那样好——这不是重点。
> 
> 至于 DYI，会有人自己打印外壳，比如修改成适合熊猫克隆版或黑熊猫等等。这种情况已经发生了。而且，如果使用定制摄像头板，你就无法 DYI 了——相比之下，网络摄像头就很容易使用了。
> 
> 至于 openpilot，目前的模型规模并不大。它们可以在配备中等 iGPU 或独立 GPU 的笔记本电脑上运行，也可以在 M 系列 MacBook 上运行，例如在最低配置的 M1 版 MacBook Air 上运行，无需将其性能发挥到极致。真的不需要什么高端的 NVIDIA GPU 或“AI 计算机”。当然，它也可以在支持 CUDA (12) 的任何设备上运行。
> 
> 更大的模型将需要专用 GPU。笔记本电脑也配备专用 GPU。
> 
> 底线是，花费[300 美元](https://discord.com/channels/469524606043160576/954493346250887168/1367962132750270574)（或更少）并在您的汽车上通过笔记本电脑运行 openpilot master。

Chinese enthusiasts have successfully run a PC version of openpilot using custom cameras and some code modifications, with the overall cost being very low.

https://github.com/user-attachments/assets/236c185c-253f-4245-af73-cc3ff046d35c

tadghh, 2026-02-07:
Not sure if this has been mentioned but you could just use the users phone, you would get both driver monitoring + front facing. Ideally the user has setup android auto/car-play. Long term maintenance of whatever app running on the device would be pretty low I imagine.

madMAx43v3r, 2026-02-27:
I have OpenGL kernels that can interpolate between images and even integrate multiple images into one. This way you can create any FPS you want.

Just need some extra compute of course, but it maps well to any GPU since it's using fixed function hardware interpolation.