commaai/openpilot #35076: Raylib stderr spam

sshane, 2025-04-26:
Most openpilot tools now display a raylib warning, can we disable it?

`RAYLIB STATIC 5.5.0.2 LOADED`

affects: juggle.py, require.py (pipeline), test_car_model.py, some pytest files:

```
batman@workstation-shane:~/openpilot$ pytest selfdrive/car/ -n32 -s -m 'not slow'
RAYLIB STATIC 5.5.0.2 LOADED
=========================================================================== test session starts ============================================================================
platform linux -- Python 3.12.4, pytest-8.3.5, pluggy-1.5.0
Using --randomly-seed=1841391711
rootdir: /home/batman/openpilot
configfile: pyproject.toml
plugins: timeout-2.3.1, xdist-3.6.1, subtests-0.14.1, mock-3.14.0, asyncio-0.25.3, cov-6.0.0, randomly-3.16.0, cpp-2.6.0, hypothesis-6.47.5, repeat-0.9.3
asyncio: mode=Mode.STRICT, asyncio_default_fixture_loop_scope=function
initialized: 10/32 workersRAYLIB STATIC 5.5.0.2 LOADED
RAYLIB STATIC 5.5.0.2 LOADED
initialized: 11/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 12/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 15/32 workersRAYLIB STATIC 5.5.0.2 LOADED
RAYLIB STATIC 5.5.0.2 LOADED
initialized: 16/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 17/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 19/32 workersRAYLIB STATIC 5.5.0.2 LOADED
RAYLIB STATIC 5.5.0.2 LOADED
initialized: 20/32 workersRAYLIB STATIC 5.5.0.2 LOADED
RAYLIB STATIC 5.5.0.2 LOADED
initialized: 21/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 22/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 23/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 24/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 25/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 27/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 28/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 29/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 30/32 workersRAYLIB STATIC 5.5.0.2 LOADED
initialized: 31/32 workersRAYLIB STATIC 5.5.0.2 LOADED
collecting: 2/32 workersRAYLIB STATIC 5.5.0.2 LOADED
```

sshane, 2025-04-26:
Maybe we can open a PR to turn this into a debug log

https://github.com/electronstudio/raylib-python-cffi/blob/56d06bdf325d19676432737fdaf697f805122daa/raylib/__init__.py#L27

closed 2025-04-26 by sshane

electronstudio, 2025-05-21:
please see https://github.com/electronstudio/raylib-python-cffi/pull/171 and then follow up  https://github.com/incognitojam/raylib-python-cffi/pull/1