commaai/openpilot #35093: macOS: can’t build after 15.4.1

greatgitsby, 2025-05-01:
### Describe the bug

tool/op.sh setup && tools/op.sh build

no longer builds. will attach stack trace. initial deep dive shows that switching to clang for failing file compilations works

### OS Version

macOS 15.4.1

### openpilot version or commit

master

phoenixs3, 2025-05-01:
This is the error i get:

`scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
gcc-13 -o opendbc_repo/opendbc/safety/tests/libsafety/safety.os -c -Wall -Wextra -Werror -nostdlib -fno-builtin -std=gnu11 -Wfatal-errors -Wno-pointer-to-int-cast -fPIC -Iopendbc_repo/opendbc/safety/tests/libsafety -Iopendbc_repo/opendbc/safety/board -Iopendbc_repo/opendbc/safety opendbc_repo/opendbc/safety/tests/libsafety/safety.c
In file included from opendbc_repo/opendbc/safety/board/fake_stm.h:2,
                 from opendbc_repo/opendbc/safety/tests/libsafety/safety.c:3:
/opt/homebrew/Cellar/gcc@13/13.3.0/lib/gcc/13/gcc/aarch64-apple-darwin24/13/include-fixed/stdio.h:75:10: fatal error: _stdio.h: No such file or directory
   75 | #include <_stdio.h>
      |          ^~~~~~~~~~
compilation terminated.
scons: *** [opendbc_repo/opendbc/safety/tests/libsafety/safety.os] Error 1
arm-none-eabi-objcopy -O binary panda/board/jungle/obj/bootstub.panda_jungle_h7.elf panda/board/jungle/obj/bootstub.panda_jungle_h7.bin
scons: building terminated because of errors.`

greatgitsby, 2025-05-01:
@phoenixs3 your issue is separate, try uninstalling `gcc@13` and `gcc-arm-embedded`, installing them again, and seeing if you see the stacktrace i attached in the description.

source: https://stackoverflow.com/a/35097817

devtekve, 2025-05-02:
FYI https://discord.com/channels/469524606043160576/1362735424644055230

devtekve, 2025-05-02:
Actually, it's 2 different issues. But they happened after 15.4. Models can't be compiled on tinygrad for me

phoenixs3, 2025-05-02:
> [@phoenixs3](https://github.com/phoenixs3) your issue is separate, try uninstalling `gcc@13` and `gcc-arm-embedded`, installing them again, and seeing if you see the stacktrace i attached in the description.
> 
> source: https://stackoverflow.com/a/35097817

I tried re-installing them with no change in the stack trace - do you know what i am doing wrong?

greatgitsby, 2025-05-02:
working build: https://github.com/commaai/openpilot/pull/35104

greatgitsby, 2025-05-13:
still two PRs outstanding to fix this, can we reopen the issue?

https://github.com/commaai/opendbc/pull/2180
https://github.com/commaai/openpilot/pull/35104

closed 2025-05-13 by adeebshihadeh