commaai/openpilot #35112: raylib spinner slow while compiling

sshane, 2025-05-03:


incognitojam, 2025-05-03:
Using pidstat it looks like there is a huge spike in CPU usage when starting the Raylib spinner, and it "idles" at 2-3% on PC. In comparison the Qt spinner load is much lower.

On device Raylib is even worse, it takes 3 seconds to initialise raylib and load the assets, and the "idle" usage is 12%.

Raylib (device)
```
$ pidstat -ruh 1 10 -e python system/ui/spinner.py

# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
11:54:30 AM  1000     54389   93.07   36.63    0.00    0.00  129.70     2   7894.06      0.00  216452   28340   0.77  python
11:54:31 AM  1000     54389   91.00    7.00    0.00    0.00   98.00     2   6114.00      0.00  331204   51576   1.40  python
11:54:32 AM  1000     54389   93.00    6.00    0.00    0.00   99.00     2   8411.00      0.00  348652   78732   2.13  python
11:54:33 AM  1000     54389   40.00    2.00    0.00    0.00   42.00     0    242.00      0.00  444380   79772   2.16  python
11:54:34 AM  1000     54389   10.00    2.00    0.00    0.00   12.00     0     19.00      0.00  444380   79772   2.16  python
11:54:35 AM  1000     54389   13.00    0.00    0.00    0.00   13.00     0     18.00      0.00  444412   79772   2.16  python
11:54:36 AM  1000     54389   12.00    0.00    0.00    0.00   12.00     0     18.00      0.00  444412   80020   2.17  python
11:54:37 AM  1000     54389   11.00    1.00    0.00    0.00   12.00     0     18.00      0.00  444412   80020   2.17  python
```

Qt (device)
```
$ pidstat -ruh 1 5 -e python common/spinner.py

# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
11:58:37 AM  1000     54462    9.90    3.96    0.00    0.00   13.86     0   2360.40      0.00   19052    8760   0.24  python
11:58:38 AM  1000     54462    0.00    0.00    0.00    0.00    0.00     0      0.00      0.00   19052    8760   0.24  python
11:58:39 AM  1000     54462    0.00    0.00    0.00    0.00    0.00     0      0.00      0.00   19052    8760   0.24  python
11:58:40 AM  1000     54462    0.00    0.00    0.00    0.00    0.00     0      0.00      0.00   19052    8760   0.24  python
11:58:41 AM  1000     54462    0.00    0.00    0.00    0.00    0.00     0      0.00      0.00   19052    8760   0.24  python
```

Raylib (PC)
```
# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
12:47:17     1000     50823  125.00    5.00    0.00    0.00  130.00     4  20293.00      0.00 1952876  231324   0.35  python
12:47:18     1000     50823    4.00    1.00    0.00    0.00    5.00     1      3.00      0.00 1952876  231324   0.35  python
12:47:19     1000     50823    4.00    1.00    0.00    0.00    5.00     1      2.00      0.00 1952876  231324   0.35  python
12:47:20     1000     50823    5.00    0.00    0.00    0.00    5.00     1      3.00      0.00 1952876  231324   0.35  python
12:47:21     1000     50823    5.00    1.00    0.00    0.00    6.00     1      3.00      0.00 1952876  231324   0.35  python
```

Qt (PC)
```
# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
12:48:12     1000     51361    1.00    0.00    0.00    0.00    1.00    12   1789.00      0.00   38652   18360   0.03  python
12:48:13     1000     51361    0.00    0.00    0.00    0.00    0.00    12      0.00      0.00   38652   18360   0.03  python
12:48:14     1000     51361    0.00    0.00    0.00    0.00    0.00    12      0.00      0.00   38652   18360   0.03  python
12:48:15     1000     51361    0.00    0.00    0.00    0.00    0.00    12      0.00      0.00   38652   18360   0.03  python
12:48:16     1000     51361    0.00    0.00    0.00    0.00    0.00    12      0.00      0.00   38652   18360   0.03  python
```

incognitojam, 2025-05-03:
This is a big help for investigating - https://plugins.jetbrains.com/plugin/16536-line-profiler

It seems every instantiation of an `rl.Vector2` or `rl.Rectangle` is really slow - we can move these out of the render loop

incognitojam, 2025-05-04:
I made a mistake by measuring `common/spinner.py` for Qt instead of the actual `_spinner` binary - its CPU is around 10-12% which makes Raylib seem less bad. I'm also making sure to run `./scripts/disable-powersave.py` because otherwise the % are much higher.

| Script | Average CPU after init |
| --- | --- |
| `rectangle.py` (with `SetTargetFPS(30)`) | 11% |
| `rectangle.py` (no `SetTargetFPS`) | 2% |
| `selfdrive/ui/_spinner` (Qt) | 11% |
| `system/ui/spinner.py` (with `SetTargetFPS(30)`) | 8-9% |
| `system/ui/spinner.py` (no `SetTargetFPS`) | 3-4% |

<details>
<summary>Full results from on device testing</summary>

```
$ pidstat -ruh 1 10 -e ./selfdrive/ui/_spinner

Linux 4.9.103 (comma-6c65f4e7)  05/04/2025      _aarch64_       (8 CPU)
Using Wayland-EGL
qt.qpa.wayland: "wl-shell" is a deprecated shell extension, prefer using "xdg-shell-v6" or "xdg-shell" if supported by the compositor by setting the environment variable QT_WAYLAND_SHELL_INTEGRATION
Using the 'wl-shell' shell integration

# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
09:30:39 PM  1000     82739   22.00   23.00    0.00    0.00   45.00     0  26589.00      0.00  361120  102348   2.77  _spinner
09:30:40 PM  1000     82739   11.00    0.00    0.00    1.00   11.00     0      1.00      0.00  361120  102348   2.77  _spinner
09:30:41 PM  1000     82739   10.89    0.00    0.00    0.00   10.89     1      0.00      0.00  361120  102348   2.77  _spinner
09:30:42 PM  1000     82739   12.00    0.00    0.00    2.00   12.00     0      0.00      0.00  361120  102348   2.77  _spinner
09:30:43 PM  1000     82739    8.00    3.00    0.00    1.00   11.00     0      0.00      0.00  361120  102348   2.77  _spinner
09:30:44 PM  1000     82739    8.00    3.00    0.00    1.00   11.00     0      0.00      0.00  361120  102348   2.77  _spinner
```

```
$ pidstat -ruh 1 10 -e python ./rectangle.py  # with SetTargetFPS(30)
Linux 4.9.103 (comma-6c65f4e7)  05/04/2025      _aarch64_       (8 CPU)
RAYLIB STATIC 5.5.0.2 LOADED
INFO: Initializing raylib 5.6-dev
INFO: Platform backend: COMMA
INFO: Supported raylib modules:
INFO:     > rcore:..... loaded (mandatory)
INFO:     > rlgl:...... loaded (mandatory)
INFO:     > rshapes:... loaded (optional)
INFO:     > rtextures:. loaded (optional)
INFO:     > rtext:..... loaded (optional)
INFO:     > rmodels:... loaded (optional)
INFO:     > raudio:.... loaded (optional)
gbm_create_device(156): Info: backend name is: msm_drm
INFO: COMMA: Using EGL version 1.5
INFO: COMMA: Found 1 valid EGL display configs
INFO: GL: Supported extensions count: 91
INFO: GL: OpenGL device information:
INFO:     > Vendor:   Qualcomm
INFO:     > Renderer: Adreno (TM) 630
INFO:     > Version:  OpenGL ES 3.2
INFO:     > GLSL:     OpenGL ES GLSL ES 3.20
INFO: GL: VAO extension detected, VAO functions loaded successfully
INFO: GL: NPOT textures extension detected, full NPOT textures supported
INFO: GL: ETC1 compressed textures supported
INFO: GL: ASTC compressed textures supported
INFO: COMMA: Initialized successfully
INFO: TEXTURE: [ID 1] Texture loaded successfully (1x1 | R8G8B8A8 | 1 mipmaps)
INFO: TEXTURE: [ID 1] Default texture loaded successfully
INFO: SHADER: [ID 1] Vertex shader compiled successfully
INFO: SHADER: [ID 2] Fragment shader compiled successfully
INFO: SHADER: [ID 3] Program shader loaded successfully
INFO: SHADER: [ID 3] Default shader loaded successfully
INFO: RLGL: Render batch vertex buffers loaded successfully in RAM (CPU)
INFO: RLGL: Render batch vertex buffers loaded successfully in VRAM (GPU)
INFO: RLGL: Default OpenGL state initialized successfully
INFO: TEXTURE: [ID 2] Texture loaded successfully (128x128 | GRAY_ALPHA | 1 mipmaps)
INFO: FONT: Default font loaded successfully (224 glyphs)
INFO: SYSTEM: Working Directory: /data/openpilot
[... 202 lines cut ...]
INFO: FONT: Data loaded successfully (120 pixel size | 95 glyphs)
INFO: FILEIO: [/data/openpilot/selfdrive/assets/fonts/Inter-Regular.ttf] File loaded successfully
INFO: TEXTURE: [ID 8] Texture loaded successfully (1024x1024 | GRAY_ALPHA | 1 mipmaps)
INFO: FONT: Data loaded successfully (120 pixel size | 95 glyphs)
INFO: FILEIO: [/data/openpilot/selfdrive/assets/fonts/Inter-SemiBold.ttf] File loaded successfully
INFO: TEXTURE: [ID 9] Texture loaded successfully (1024x1024 | GRAY_ALPHA | 1 mipmaps)
INFO: FONT: Data loaded successfully (120 pixel size | 95 glyphs)
INFO: FILEIO: [/data/openpilot/selfdrive/assets/fonts/Inter-Thin.ttf] File loaded successfully
INFO: TEXTURE: [ID 10] Texture loaded successfully (1024x1024 | GRAY_ALPHA | 1 mipmaps)
INFO: FONT: Data loaded successfully (120 pixel size | 95 glyphs)
INFO: FILEIO: [/data/openpilot/selfdrive/assets/img_spinner_comma.png] File loaded successfully
INFO: IMAGE: Data loaded successfully (848x848 | R8G8B8A8 | 1 mipmaps)
INFO: TEXTURE: [ID 11] Texture loaded successfully (360x360 | R8G8B8A8 | 1 mipmaps)
INFO: FILEIO: [/data/openpilot/selfdrive/assets/img_spinner_track.png] File loaded successfully
INFO: IMAGE: Data loaded successfully (848x848 | R8G8B8A8 | 1 mipmaps)
INFO: TEXTURE: [ID 12] Texture loaded successfully (360x360 | R8G8B8A8 | 1 mipmaps)

# Time        UID       PID    %usr %system  %guest   %wait    %CPU   CPU  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
09:42:06 PM  1000     82939   78.00   12.00    0.00    0.00   90.00     0  11365.00      0.00  608388   79928   2.16  python
09:42:07 PM  1000     82939    3.00    1.00    0.00    0.00    4.00     0     19.00      0.00  608388   79928   2.16  python
09:42:08 PM  1000     82939    3.00    0.00    0.00    0.00    3.00     0     18.00      0.00  608388   79928   2.16  python
09:42:09 PM  1000     82939    3.00    1.00    0.00    0.00    4.00     0     18.00      0.00  608420   80188   2.17  python
09:42:10 PM  1000     82939    2.00    2.00    0.00    0.00    4.00     0     18.00      0.00  608420   80188   2.17  python
```
</details>

incognitojam, 2025-05-04:
I have seen the spinner lag before, but I can't repro it by starting processes to use CPU - even with 100% cpu activity the spinner is managing 30fps. Could it be GPU related when doing the model compile with tinygrad?

[image]

sshane, 2025-05-04:
Have you scons cleaned and reboot?

sshane, 2025-05-08:
From a clean scons it completely froze 

https://github.com/user-attachments/assets/101fc67f-aeca-4b1a-aea3-749d91f5c721

sshane, 2025-05-08:
When I unplugged it after maybe 10 mins, it resumed near the end so it likely was still compiling?

deanlee, 2025-05-08:
I’m unable to reproduce this issue on my device. @incognitojam : Could it be related to a timing issue? The latest code uses `rl.get_frame_time()` to adjust the rotation angle.

sshane, 2025-06-06:
this still happens

closed 2025-06-07 by sshane