commaai/openpilot #35160: Comma drives into left lane

Edamame25, 2025-05-08:
### Describe the bug

Comma drives over lane line into the left lane on the majority of drives. The display shows the green driving path going onto the lane line. 

NOTE: Comma operates fine and hands-free on SR29 from Lakeport to the intersection of SR29 and SR20 and back. This is a newly painted and freshly paved highway. 

Slightly improved (requiring less intervention, hands free for ~75% of the drive) on US101 from Geyserville to San Rafael. However, US101 from San Rafael to Geyserville requires constant intervention.

On the majority of Northern CA and Bay Area freeways, highways, and city streets, it requires full hands on the wheel to keep the car pulled to the right. Really prominent with yellow lines and double yellow lines--when driving without Comma on, the car can be centered in the lane and as soon as Comma is turned on, it will pull the car over to the left onto the yellow lines. 

Chevy Bolt EUV Premier 2023 with ACC. Comma is measured centered and wheels are aligned. (Occurs on Open Pilot, Frog Pilot, and Sunny Pilot. Comma support suggested to post the issue here.) 

### Provide a route where the issue occurs

00000000--178d387436--7

### openpilot version

0.9.8

sshane, 2025-05-13:
What is your dongle id?

Edamame25, 2025-05-13:
Hi. I just updated the dongle/route in the bug report: 86dcf894ec79a156/00000001--d1708114ef

Edamame25, 2025-05-14:
[image]

adeebshihadeh, 2025-09-01:
Still having issues?