commaai/openpilot #35164: Deceleration stopped working on latest

mvl-boston, 2025-05-08:
### Describe the bug

Deceleration is not working as of commit 33849245d8025786c97881a8c132989023bb62b7 on master.  Comma sees the lead car with the yellow triangle onscreen, but does not slow down.

Details: https://discord.com/channels/469524606043160576/1369659680585941046/1369659680585941046

Thinking this was an issue with the TR7 upgrade, I tried commit 36ff474bc812474b1fd15b1c5c58252563034fea and was still unsuccessful.   I had previously been using master as of May 3, so the issue started sometime after that date.

### Provide a route where the issue occurs

9662a1dab70c9dc3/0000005f--db363d96c4

### openpilot version

0.9.9 33849245d8025786c97881a8c132989023bb62b7 

### Additional info

Test using 36ff474bc812474b1fd15b1c5c58252563034fea:
ad9840558640c31d/0000002f--edbdc09353

adeebshihadeh, 2025-05-08:
We can't look into issues that happen on forks (even if it's just car support).

closed 2025-05-08 by adeebshihadeh

mvl-boston, 2025-05-08:
Thanks, I've asked Honda discord for someone with an upstreamed car to test.