commaai/openpilot #35168: athena: recently viewed routes no longer works

sshane, 2025-05-10:


fivemtebex, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

fivemeyestore, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

scriptforfivem, 2025-05-10:
I'm impressed by the modularity of this codebase. Makes it very maintainable.

fivemserverhostings, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

fivemhostingserver, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

fivemeyestore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemserverhostings, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

eyestorefivem, 2025-05-10:
The documentation in this repository is excellent! Very clear and comprehensive.

scriptforfivem, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyesgodcode, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

fivemservereyestore, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

godofcodeeyes, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

myeyestore, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

adeebshihadeh, 2025-05-13:
Sounds good

sshane, 2025-05-23:
fixed once connect gets deployed

closed 2025-05-23 by sshane