commaai/openpilot #35228: kernel panic when setting ignition on and off repeatedly with a jungle

maxime-desroches, 2025-05-14:
Easy to reproduce with a jungle by pressing the ignition button quickly and repeatedly.

Full log attached: [kernel_panic.txt](https://github.com/user-attachments/files/20214636/kernel_panic.txt)

BBBmau, 2025-05-14:
is the panic only on comma 3's? I don't see a 3X label.

maxime-desroches, 2025-05-15:
No, panics on both 3X and 3

deanlee, 2025-05-15:
It’s hard to trace the entire process through the logs. The crash seems to have occurred  while managing RT task throttling. 

1. the `selfdrive.model`  consuming too much CPU, triggering RT throttling:
  ```
  <3>[  251.856293] sched: RT throttling activated for rt_rq         pK-error (cpu 7)
  <3>[  251.856293] rt_period_timer: expires=251870132126 now=251860026864 period=1000000000
  <3>[  251.856293] potential CPU hogs:
  <3>[  251.856293] current selfdrive.model (52727) is running for 505490679 nsec
  <3>[  251.856293] 	selfdrive.model (52727)
  <3>[  251.856293] 	selfdrive.model (52799)
  <4>[  251.856307] ------------[ cut here ]------------
  <2>[  251.856308] Kernel BUG at ffffff9e1f5043f8 [verbose debug info unavailable]
  ```
2. When the kernel attempted to throttle the task and log details, it crashed in the `dump_throttled_rt_tasks` , possibly due to a bug:
  ```
  dump_throttled_rt_tasks+0x1a0/0x1e8
  task_tick_rt+0x28/0x138
  scheduler_tick+0xac/0x208
  update_process_times+0x60/0x78
  tick_sched_handle.isra.5+0x38/0x70
  tick_sched_timer+0x54/0x190
  __hrtimer_run_queues+0x18c/0x380
  hrtimer_interrupt+0xa8/0x1d8
  ```

deanlee, 2025-05-15:
looks like  there are two instances of the `selfdrive.model` process running simultaneously

> <3>[  251.856293] 	selfdrive.model (52727)
> <3>[  251.856293] 	selfdrive.model (52799)

closed 2026-08-16 by adeebshihadeh