commaai/openpilot #35237: Ford: openpilot unavailable on “L” gear.

OvenBaked360, 2025-05-15:
I have a 2021 ford escape hybrid and am unable to us the L gear while using open pilot which has cost me about 4 mpg.  In L mode / gear it does not change the gear ratios. It changes the amount of regenerative braking and does a sudo one pedal drive mode. I was instructed to disable open pilot do a drive using factory ACC and upload a ticket here by ssshaneee on Reddit. Here is the link from comma connect. 
https://connect.comma.ai/457e84b882411077/00000031--31dbc709aa

sshane, 2025-05-15:
@OvenBaked360 can you try this branch out? Once you do and report back with a route, we can merge! https://github.com/commaai/openpilot/pull/35238

Installer URL:

https://installer.comma.ai/commaai/ford-low

sshane, 2025-05-15:
Can you also put your device on WiFi to upload logs?

closed 2025-05-18 by sshane