commaai/openpilot #35339: Calibratable Steering Torque Multiplier necessary for Rivian

MrMusAddict, 2025-05-24:
Rivian Gen1 comes with its own driver assistance called Driver+ which controls steering on pre-mapped highways. The torque available to Driver+ is in my experience 2-3x more powerful than the maximum that OpenPilot offers. There are curves on I5 that Driver+ not only handles effortlessly, but requires a small effort to pull the steering wheel away for a disengagement.

In those same curves on I5, OpenPilot peters out and requires driver intervention. However other drivers with brand new tires can handle the same curves just fine. As I understand from Discord, the **_absolute_** torque performance is consistent across all Rivian vehicles, but due to tire wear this causes **_universal inconsistency_**.

I propose that OpenPilot add in the ability to calibrate this inconsistency out. I imagine that could be some kind of Torque Multiplier that gets adjusted (within a reasonable range) via calibration.

MrMusAddict, 2025-05-26:
Specifically for rivian (but also maybe any other occurrences where there's discrepancy within a single driving platform), would it be possible to use occurrences of the "speed too fast for turn" warning to identify vehicles that need a boost?

For example, there are turns on I5 that receive this error for me despite me only going 68 MPH, and they're really tame turns. Talking around 0.1 lateral G's, and it's still floating out of the lane.

Could these become identifiable/verifiable triggers to give Max Accel a small boost?

sshane, 2025-05-27:
Can you provide routes?

MrMusAddict, 2025-05-27:
I'll keep you posted. Looks like the routes with them cleared out of my history on connect. Also I've been assisting it with MADS on SunnyPilot so there have been less warnings (since I've slowed the vehicle on some turns using 1PD). However this ticket was created from observed behavior on stock OP.

sshane, 2025-06-30:
We will need routes to take action here, please report back if you can send some.