commaai/openpilot #35475: Spurious? (or maybe real?) `release3-staging` → "Fan Malfunction: Likely Hardware Issue"

yuzisee, 2025-06-06:
### Describe the bug

Only after switching to release3-staging did I start seeing "Fan Malfunction: Likely Hardware Issue" on my device.

Maybe it's a coincidence and there is actually a hardware issue. Or, maybe there's a regression somewhere between 0.9.8 and 0.9.9 that's causing these alerts incorrectly?

Anyway, better to report than not report I suppose, since we're in the `release3-staging` process after all.

### Provide a route where the issue occurs

21015c36062c3cef/00000032--c480bfafc0

### openpilot version

release3-staging (for 0.9.9)

### Additional info

The alert does seem to be intermittent, and went away on its own after a while. I've seen this a few times now and every time it eventually resolves itself without me doing anything.

My assumption is that if there were an actual fan failure the alert would stay up & my device would overheat... but neither of those things have happened yet? Nonetheless, fans are finnicky things so it's hard for me to tell.

adeebshihadeh, 2025-06-06:
Here's what happened in that drive. I think it's most likely your hardware starting to fail, but I'll check for a regression too.

[image]

adeebshihadeh, 2025-06-06:
I'm not seeing anything in the diff that would cause this. Want to try running current release3 for a bit to see if it still happens?

adeebshihadeh, 2025-06-08:
I see the issue still happens on release3, so guess we'll close this? 

[image]

yuzisee, 2025-06-08:
> I see the issue still happens on release3, so guess we'll close this? 
> 
> [image]

Hmm, I haven't seen a single "Fan Malfunction: Likely Hardware Issue" pop up since switching to `release3`, but if it looks like the fan is doing the same thing -- was there a chance in what triggers the popup?

adeebshihadeh, 2025-06-10:
Want to give this branch a try: `installer.comma.ai/commaai/revert-fan`?

closed 2025-06-17 by adeebshihadeh

adeebshihadeh, 2025-07-03:
FYI now that 0.9.9 has been in release for a bit, I can confirm there was no uptick of fan malfunctions.