commaai/openpilot #35592: Request for Support: 2024 Jeep Wagoneer (WS Platform)

cmurdoc, 2025-06-23:
Body:
Vehicle: 2024 Jeep Wagoneer (WS Platform)
Trim: Series II
Engine/Drivetrain: 3.0L Hurricane Twin Turbo, 4x4

I’d like to request support for the 2024 Jeep Wagoneer L in openpilot. This vehicle uses Stellantis’ WS platform and includes:
	•	Full-speed Adaptive Cruise Control with Stop-and-Go
	•	Front-facing radar and camera
	•	Active Lane Management system

I own the vehicle and am willing to assist with reverse-engineering and data collection.

Support Evidence
	•	Vehicle uses radar + camera for ACC (same as other FCA vehicles supported by openpilot)
	•	Steering is electrically assisted
	•	CAN bus architecture appears similar to other recent Jeep platforms (e.g., WL75 Grand Cherokee L)

I’m comfortable capturing cabana drives, fingerprinting ECUs, and probing CAN signals with my comma 3X (that I currently use for my Telluride).
Please let me know what traces, logs, or DBC-style details are most useful to begin.

ElKozlo, 2025-06-27:
Owner of a 2024 Jeep wagoneer here as well looking for the same thing. Willing to assist in any way I can as well.

adeebshihadeh, 2025-07-12:
Closing as this isn't a bug - check out our [Discord](https://discord.comma.ai) instead.

closed 2025-07-12 by adeebshihadeh

cmurdoc, 2025-07-13:
Can you please provide instructions on how to go about getting my car supported? Telling me to go to the discord channel, which I'm a part of, is not helpful.

iReadTOS, 2025-09-02:
@cmurdoc 
Hi all,
My dad owns a Wagoneer L and he has long commutes everyday. I would love to try and get the Comma as a Christmas gift or something.

I was doing research and found this:
https://github.com/commaai/opendbc#how-to-port-a-car

If the author of issue is still available, I hope this can help! I cannot really afford to buy a ton of developer tools or similar right now, but I still would love to be able to assist in anyway, shape, or form as needed.

cmurdoc, 2025-09-02:
Hi @iReadTOS ,
I just purchased the wiring harness compatible with the Jeep Grand Cherokee. After removing the windshield camera assembly cover and comparing the part numbers between the Wagoneer and the Cherokee, there does appear to be identical components used in all of these vehicles. I'll have more clarity once I receive this harness. There is a tiny chance that 1.) It just plugs in and 2.) The comma 3x fingerprints the Wagoneer as a Grand Cherokee and "just works". Praying that this is the case!

m-alawar, 2025-11-07:
> Hi [@iReadTOS](https://github.com/iReadTOS) , I just purchased the wiring harness compatible with the Jeep Grand Cherokee. After removing the windshield camera assembly cover and comparing the part [...]

Curious to hear updates on this

I have a 2023 GW and open pilot would just make it perfect