commaai/openpilot #35774: Dashcam mode - Car unrecognized

gcantrell82, 2025-07-20:
long story short device has worked amazing for me until recently. Have been a stock software guy since I got it until I saw this message a few weeks ago, did some looking around online saw where switching to nightlys could fix it and it did for about a week or so and now it’s back to this message. Is there another way to fix and are there ways to prevent from getting in this predicament? 

[image]

adeebshihadeh, 2025-07-20:
What's your device's dongle ID?

gcantrell82, 2025-07-20:
How do I fill in a route if there are no routes in my account? I don’t subscribe to connect 

It’s preventing me from submitting the ticket. 

[image]
[image]