commaai/openpilot #35779: Driver Monitor: Low Sensitivity to Eye Closure

Jiacheng-Liu, 2025-07-21:
Issue Description:

  The driver monitoring system appears to rely heavily on head position
  and orientation, while being largely insensitive to whether the
  driver's eyes are open or closed. I've noticed that as long as my head
   remains in a forward-facing position, the system does not
  consistently register eye closure as a sign of distraction.

  Observed Behavior:

  I can keep my head perfectly still and facing forward, but close my
  eyes for several seconds without triggering a driver attentiveness
  warning. The system only seems to react reliably to head movements
  (looking away, looking down, etc.).

  Expected Behavior:

  The driver monitor should be able to reliably detect when the driver's
   eyes are closed for a prolonged period (e.g., > 2-3 seconds) and
  issue a distraction warning, regardless of head position. This is a
  critical safety feature for detecting microsleeps or a driver falling
  asleep.

jasseeeem, 2025-08-31:
I tested Driver Monitor with few images and here are few observations that could help here:
* `rightBlinkProb` and `leftBlinkProb` are very high (**as expected**) when both the eyes are closed.
* When one of the eyes are closed, both `rightBlinkProb` and `leftBlinkProb` are always close to 0 (**which is unexpected**).
*

ZwX1616, 2026-04-23:
we are actively working on this - the current master model should already see some improvements

closed 2026-04-23 by ZwX1616