commaai/openpilot #35788: Kernel panic

sshane, 2025-07-22:
My comma 3X just froze and reboot after parking. I first noticed it by my car throwing a fault on the dash since I guess the screen froze too (didn't show any alerts, went to black after a few seconds).

https://connect.comma.ai/d9b97c1d3b8c39b2/00000164--129ea9d3d4

After reboot: https://connect.comma.ai/d9b97c1d3b8c39b2/00000165--1cfbe9210b

few kernel oops/bugs in the bootlog: 000000d1--254ca472f6.zst

adeebshihadeh, 2025-07-31:
Confirmed it's not a regression so removing from the milestone

jyoung8607, 2026-02-05:
Two more incidents here: https://github.com/commaai/openpilot/issues/37044#issuecomment-3856257841

MarcoTheDingo, 2026-02-20:
Hey all. I can somewhat reliably recreate this by driving away from my home network, going off-road, then driving home and staying in the driveway, where the Wi-Fi is slightly weaker while staying on-road. It does also happen in other scenarios when I'm connecting to Wi-Fi while on-road, but that is the most reliable to recreate.

I've been having this issue for months, both on stock OP and sunnypilot. Let me know if there's anything I can do to assist with this. My dongle id is a2152effbad22b70.

suburbazine, 2026-03-05:
Had an onroad crash that may be related to this:
https://connect.comma.ai/208f95b6156cdcad/00000027--f408136cec/833/1141

zappybiby, 2026-03-29:
I've only ever used Sunnypilot dev, so I apologize if my report is not valid for this OP discussion, but I have experienced this (or a related) WLAN error repeatedly and wanted to provide my log. I have only experienced this issue specifically when I am connected to my phone's mobile hotspot and the network is set to 'untethered'. Although the time it takes to fault is variable, I can almost always inevitably get it to happen after enough time with those two conditions. I have not encountered this issue when the connection is metered (which I believe means that it won't upload to Comma Connect). 

```text
device      mici (Comma 4), dongle_id=9f7170b0e093c614
car         subaru / SUBARU_IMPREZA_2020
software    sunnypilot dev, initData.version=2026.001.000, ctx.version=2026.03.18-4343
commit      d12485cde21885622de3a4e0dd307040065ccc78 ('1773839418 2026-03-18 13:10:18 +0000')
```

State at time of fault:
- Driving without Sunnypilot engaged.
- Connected to phone's Mobile Hotspot, unmetered connection
- My mobile hotspot is the only network I've ever connected to. 

## Crash / fault

This is about `4.04 seconds` after the last healthy retained `carState` at `2026-03-29 11:28:26.697 AM EDT`.  While driving with Sunnypilot not engaged, my Comma 4 suddenly reboots and my stock ACC (Eyesight) is permanently disabled until my car is next restarted.

<details>
<summary>pstore / kernel crash log excerpt</summary>

```text
[ 3362.473508] arm-smmu 15000000.apps-smmu: Unhandled context fault: iova=0xaa6eb0d8, fsr=0x40000402, fsynr=0x360003, cb=5
[ 3362.473544] arm-smmu 15000000.apps-smmu: FAR    = 00000000aa6eb0d8
[ 3362.473565] arm-smmu 15000000.apps-smmu: FSR    = 40000402 [TF SS ]
[ 3362.473586] arm-smmu 15000000.apps-smmu: soft iova-to-phys=0x0000000000000000
[ 3362.473606] arm-smmu 15000000.apps-smmu: SOFTWARE TABLE WALK FAILED! Looks like 15000000.apps-smmu accessed an unmapped address!
[ 3362.473625] arm-smmu 15000000.apps-smmu: hard iova-to-phys (ATOS) failed
[ 3362.473644] arm-smmu 15000000.apps-smmu: SID=0x40
[ 3362.473662] arm-smmu 15000000.apps-smmu: Unhandled arm-smmu context fault!
[ 3362.473707] Kernel BUG at ffffff849147fc68 [verbose debug info unavailable]
[ 3362.473732] ------------[ cut here ]------------
[ 3362.473745] Kernel BUG at ffffff849147fc68 [verbose debug info unavailable]
[ 3362.473758] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[ 3362.473799] CPU: 1 PID: 201 Comm: irq/170-arm-smm Not tainted 4.9.103 #5
[ 3362.473811] Hardware name: comma mici (DT)
[ 3362.473825] task: ffffffecb26ce200 task.stack: ffffffecb1b88000
[ 3362.473846] PC is at arm_smmu_context_fault+0x5a0/0x5a8
[ 3362.473860] LR is at arm_smmu_context_fault+0x5a0/0x5a8
[ 3362.473873] pc : [<ffffff849147fc68>] lr : [<ffffff849147fc68>] pstate: 40c00145
[ 3362.473886] sp : ffffffecb1b8bcb0
[ 3362.473898] x29: ffffffecb1b8bcd0 x28: ffffff80100c5058
[ 3362.473927] x27: ffffffecb20f3400 x26: ffffff8000000040
[ 3362.473956] x25: 0000000000000000 x24: 00000000aa6eb0d8
[ 3362.473985] x23: ffffff80100c5000 x22: 0000000040000402
[ 3362.474014] x21: ffffffecb20f3530 x20: ffffff8493738788
[ 3362.474043] x19: ffffffecb3673018 x18: 0000000000000010
[ 3362.474072] x17: 0000007fae9b61e0 x16: ffffff8490f3f7e0
[ 3362.474101] x15: ffffffffffffffff x14: 6620747865746e6f
[ 3362.474130] x13: 6320756d6d732d6d x12: 72612064656c646e
[ 3362.474158] x11: 0000000000002bff x10: 6d732d737070612e
[ 3362.474187] x9 : 0000000000003ff0 x8 : 0000000000003fff
[ 3362.474216] x7 : ffffffecbdc54090 x6 : 0000000005052655
[ 3362.474245] x5 : 00ffffffffffffff x4 : ffffff8493739000
[ 3362.474273] x3 : ffffff8493739628 x2 : 0d68e193b0e22f00
[ 3362.474302] x1 : 0d68e193b0e22f00 x0 : 0000000000000000
[ 3362.475487] Process irq/170-arm-smm (pid: 201, stack limit = 0xffffffecb1b88000)
[ 3362.475647] icnss: Received force error fatal request from FW
[ 3362.551780] Fixing recursive fault but reboot is needed!
[ 3362.627895] Fatal error on the modem.
[ 3362.627946] modem subsystem failure reason: err_qdi.c:456:EF:wlan_process:1:cmnos_thread.c:3900:Asserted in whal_interrupt_wifi_ip02.c:whalGetPendingInterrupts:499.
[ 3362.627954] subsys-restart: subsystem_restart_dev(): Restart sequence requested for modem, restart_level = RELATED.
[ 3362.630723] ipa-wan ipa3_ssr_notifier_cb:2692 IPA received MPSS BEFORE_SHUTDOWN
[ 3362.654716] rmnet_ipa started deinitialization
[ 3362.654759] ipa-wan ipa3_wwan_remove:2516 Failed to teardown IPA->APPS pipe
[ 3362.654802] ipa-wan ipa3_wwan_remove:2521 Failed to teardown APPS->IPA pipe
[ 3362.729627] ipa-wan ipa3_ssr_notifier_cb:2708 IPA BEFORE_SHUTDOWN handling is complete
[ 3362.729664] icnss: PD service down, pd_state: 2, state: 0x4d8f: cause: Root PD crashed
[ 3362.729903] icnss: ASSERT at line 2460
[ 3362.729941] ------------[ cut here ]------------
[ 3362.729955] Kernel BUG at ffffff84913d1e3c [verbose debug info unavailable]
[ 3362.730006] Internal error: Oops - BUG: 0 [#3] PREEMPT SMP
[ 3362.730060] CPU: 1 PID: 63751 Comm: kworker/u16:3 Tainted: G      D         4.9.103 #5
[ 3362.730087] Workqueue: icnss_driver_event icnss_driver_event_work
[ 3362.730123] PC is at icnss_driver_event_work+0x131c/0x18d0
```

</details>

My dump appears near-identical to the one posted at https://github.com/commaai/openpilot/issues/35788#issuecomment-3864730638. 
[... cut ...]
rlogs of the two segments preceding the fault, the boot log with the fault, and the first segment following the fault.
[logs_wlan_crash_bundle_20260329.zip](https://github.com/user-attachments/files/26335316/logs_wlan_crash_bundle_20260329.zip)

```
time periods covered:
  -before-fault-seg12.zst: 11:27:18.742 AM to 11:28:18.796 AM EDT
  -before-fault-seg13.zst: 11:28:18.738 AM to 11:28:26.661 AM EDT
  -fault-bootlog.zst: crash sequence is visible at about 11:28:30.581 AM to 11:28:30.878 AM EDT
  -post-fault-seg0.zst: 11:29:22.716 AM to 11:30:23.706 AM EDT
```

zappybiby, 2026-04-03:
The way I reproduce this issue best for debugging is to connect my phone to my home wifi (wifi 5 for the record), then turn on mobile hotspot (sharing the home wifi through my phone).  I then connect Comma, set network to 'unmetered'. With this method I was able to trigger the fault 7 times today, typically taking less than 3 minutes from boot to trigger the fault. I usually don't need to do anything further after it connects to my mobile hotspot, but if it's taking too long, I just start uploading routes using Comma Connect and usually shorter after it causes the SSMU fault. There are times where I still can't get it to trigger this way, idk.

In my debug build of the kernel, here's what i've gathered

1. The faulting IOVA (they are all very similar), 0xa86eb0c8 is unmapped, and the follow-on faults hit nearby addresses, it's walking a bad descriptor/ring structure rather than one isolated access.

- DTS shows ICNSS has a general SMMU window at `0xa0000000-0xafffffff` and a separate IPA window at `0xb0000000-0xb000ffff`; this fault is in the general window, not the IPA window.

- The faulting address lands in ICNSS’s main general-use SMMU range (0xa0000000–0xafffffff), not in the separate IPA (IP Accelerator) range (0xb0000000–0xb000ffff).

- HTC, the host↔firmware transport layer, looks healthy in the logs. Credit flow is normal, queue depth stays at 1 or less, and the last successful credit activity happens immediately before the failure.

- The WMI command queue looks normal. It is mostly sending WMI_REQUEST_STATS_CMDID requests, and the 0x10d payload matches the normal stats-request bitmap. 

2. The connectivity timeline does not show a roam/connect transition near any of my recorded crashes. I monitored the following events: `connect_req, disconnect_req, scan_req, scan_done, roam, conn_state, auth_state, and connect_result`  and found that the connection was healthy, there were no recent disconnects or abnormal connection problems leading up to or at the time of any of the crashes. I feel this is important because while it has been mentioned to be a factor in this fault in other reports, it does not appear to be a necessary cause to induce the crash. 

3. In some logs, a file upload had just started within a second of the ssmu fault prior, but in other logs, it had been over 10 seconds without a file being uploaded / upload initiated (note: I don't have a great 'monitor ALL network activity, Tx etc etc across the board' hookup so I can't say much more than this). I didn't see any obvious upload failing or related problems in my logs.

<details>
<summary>Timeline of one crash</summary>
**Fault Timeline Relative To SMMU Fault**

`WLAN diag monitors and logs the following events:  connect_req, disconnect_req, scan_req, scan_done, roam, conn_state, auth_state, and connect_result` 

#### `T-201.788s`
- `WLAN diag[1]` `roam` `roam_status=25`, `roam_result=0`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-201.788s`
- `WLAN diag[2]` `conn_state old=not_connected new=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-201.707s`
- `WLAN diag[3]` `scan_req` `scan_id=40961`, `source=vendor`, `n_channels=36`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-199.330s`
- `WLAN diag[4]` `scan_done` `scan_id=40961`, `source=vendor`, `status=0`, `aborted=0`, `age_ms=2377`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-198.275s`
- `WLAN diag[5]` `scan_req` `scan_id=40962`, `source=vendor`, `n_channels=36`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-195.987s`
- `WLAN diag[6]` `scan_done` `scan_id=40962`, `source=vendor`, `status=0`, `aborted=0`, `age_ms=2288`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-193.273s`
- `WLAN diag[7]` `scan_req` `scan_id=40963`, `source=vendor`, `n_channels=36`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-191.073s`
- `WLAN diag[8]` `scan_done` `scan_id=40963`, `source=vendor`, `status=0`, `aborted=0`, `age_ms=2200`, `state=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-191.008s`
- `WLAN diag[9]` `connect_req` for `Brian's S24+`, `bssid=52:f5:88:f9:29:ca`. Buffered WLAN history from `ts=52270165083ns`.
- Source: `console-ramoops-0`

#### `T-191.008s`
- `WLAN diag[10]` `conn_state old=not_connected new=not_connected`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-191.005s`
- `WLAN diag[11]` `conn_state old=not_connected new=connecting`. Buffered WLAN history.
- Source: `console-ramoops-0`

#### `T-191.004s`
- `WLAN diag[12]` `roam` `roam_status=10`, `roam_result=0`, `state=connecting`. Buffered WLAN history.
[... 139 lines cut ...]
- `-5.873s` `swaglog.0000002006:1` `athena.jsonrpc_handler.call_method`
  `method=listUploadQueue id=b04890b1-7c89-45cc-953d-03454e173bfe`

- `-5.838s` `swaglog.0000002006:2` `athena.jsonrpc_handler.call_method`
  `method=getNetworkMetered id=4d46a7b6-a813-4f8f-8313-f4644f0d7561`

- `-4.033s` `swaglog.0000002006:3` `athena.upload_handler.success`
  `fn=11/rlog.zst sz=10373137 network_type=1 metered=false`

- `-4.020s` `swaglog.0000002006:4` `athena.upload_handler.upload_start`
  `fn=13/fcamera.hevc sz=37479163 network_type=1 metered=false retry_count=0`

- `-2.578s` `swaglog.0000002006:5` `athena.jsonrpc_handler.call_method`
[... cut ...] 5dcdd915`

- `-2.576s` `swaglog.0000002006:6` `athena.jsonrpc_handler.call_method`
  `method=listUploadQueue id=0f248915-9c55-42d3-aecc-51f6286bffcc`
```

</details>

i'm investigating if maybe its a fault walking HTT RX paddr ring,, 8-byte ofsets 

This is my big dump I have not had time to make anything pretty, this contains most of the main diags so far from one of my logs, I'm still working on this.
[dmesg-ramoops-0.txt](https://github.com/user-attachments/files/26472556/dmesg-ramoops-0.txt)

robbederks, 2026-04-24:
@zappybiby thanks for looking into it! i've been trying to reproduce it without success for a while, and even using my phone hotspot now doesn't seem to do the trick. a few questions:

- did you set your phone's SSID to the same as your home network? if not, do you have saved connections for both on your device such that it tries to roam between them?
- could you quickly check which wlan channels the networks are on?
- in case it matters, what phone and APs do you have?
- anything else that might not be obvious to reproduce it?