commaai/openpilot #35803: DM: unexpected entry into tooDistracted state

jyoung8607, 2025-07-24:
### Describe the bug

On a recent road trip I tried out the always-on DM feature for the first time. On this same trip I got put in the `tooDistracted` penalty box on two different drives. I have *never* been in the penalty box before and I don't understand how I got there. I drove engaged normally, disengaged, and then could not reengage.

I make no claim to driver attention sainthood, and perhaps correlation ~= causation with the always-on DM toggle, but I don't understand how I got into the penalty box without a bunch of bright red full screen DM warnings with loud obnoxious sounds first. This feels like a bug.

### Provide a route where the issue occurs

`f39cf149898833ff/0000000c--64c8f895cd` and one other I'll dig up if needed

### openpilot version

d02f78a59caa5a5734e5ce10893c2fbdc1cef859 with some VW controls experiments

adeebshihadeh, 2025-07-24:
I also happened to hit this for the first time today.
```
84ec13cd18bc76ca/00000221--36c87d4fa6
```

simontheflutist, 2025-07-24:
related to? https://github.com/commaai/openpilot/issues/35776

jyoung8607, 2025-07-29:
Resolved by #35834

closed 2025-07-29 by jyoung8607