commaai/openpilot #35842: NVMe Drive Permissions Error on Bootup.

GameGrime, 2025-07-29:
### Describe the bug

Looks to be a bug with the NVMe SSD handling on the recent release branch.
Compounded with a lack of error transparency.

Currently on startup it is possible for the SSD permissions to error and write permissions are lost to the drive. Causing loggerd to be unable to start. This results in no logs being stored of the issue _because you can't write to the SSD to store the logs._

There has been references to this in #34742 and #35473 but these closed issues does not fix the root of the problem.

Right now there is no feedback as to when the error occurs. There is no backup writing location for logs when the NVMe drive is unwriteable. 

A common "solution" is to **reboot/restart** the device. This does not work consistently. Some attempts rebooting the device allows the SSD to be writeable on the next bootup. Other times there is no solution. I was able to reboot 15 times in a row one day without the error fixing itself. 3 hours later, it started up just fine.

### Provide a route where the issue occurs

af428589210e270a

### openpilot version

0.9.9

jyoung8607, 2025-07-30:
The [errors you found](https://github.com/commaai/openpilot/issues/35473#issuecomment-3096818909) indicate the data partition is mounted readonly. That's not really a permissions issue. It's more likely the SSD filesystem is corrupted or otherwise in a state where RW mounting isn't possible.

The other PRs don't magically fix any SSD hardware issue that may exist, they just make it so SSD problems that happen onroad don't force an immediate disengagement.

Since you can reproduce this easily, it would be interesting to gather some info while it's in this state. Since the output is long, we'll send the output to termbin so all you have to do is copy the links.

```
sudo tune2fs -l /dev/nvme0n1 | nc termbin.com 9999
sudo smartctl -a /dev/nvme0 | nc termbin.com 9999
```

These commands will show us any filesystem corruption, and should fix you up at least temporarily.

```
sudo umount /data/media
sudo fsck -y /data/media | nc termbin.com 9999
sudo mount /data/media
```

To be clear, there's a good chance you have a real SSD problem, but it's worth exploring ways that openpilot might better alert and recover when this sort of issue is present.

adeebshihadeh, 2025-08-28:
Closing as comma three is no longer supported on openpilot master.

closed 2025-08-28 by adeebshihadeh

GameGrime, 2025-08-28:
Ok.

For anyone finding this thread in the future. I took out my SSD from the Comma 3 and it starts up with no issue.