commaai/openpilot #35855: address in used and seg fault on Macbook

josephykwang, 2025-07-31:
### Describe the bug

``` tools/replay/replay --demo    ``` will either get ```active services: accelerometer, accelerometer2, alertDebug, androidLog, cameraOdometry, can, carControl, carOutput, carParams, carState, clocks, controlsState, customReservedRawData0, customReservedRawData1, customReservedRawData2, deviceState, driverAssistance, driverCameraState, driverEncodeData, driverEncodeIdx, driverMonitoringState, driverStateV2, drivingModelData, errorLogMessage, gnssMeasurements, gpsLocation, gpsLocationExternal, gpsNMEA, gyroscope, gyroscope2, lightSensor, liveCalibration, liveDelay, liveParameters, livePose, liveTorqueParameters, liveTracks, livestreamDriverEncodeData, livestreamDriverEncodeIdx, livestreamRoadEncodeData, livestreamRoadEncodeIdx, livestreamWideRoadEncodeData, livestreamWideRoadEncodeIdx, logMessage, longitudinalPlan, magnetometer, managerState, modelV2, navInstruction, navRoute, navThumbnail, onroadEvents, pandaStates, peripheralState, procLog, qRoadEncodeData, qRoadEncodeIdx, qcomGnss, radarState, rawAudioData, roadCameraState, roadEncodeData, roadEncodeIdx, selfdriveState, sendcan, soundPressure, temperatureSensor, temperatureSensor2, testJoystick, thumbnail, touch, ubloxGnss, ubloxRaw, uploaderState, wideRoadCameraState, wideRoadEncodeData, wideRoadEncodeIdx
                      Error, failed to bind PubSocket to gyroscope: Address already in use
                                                                                          Assertion failed: (socket), function PubMaster, file socketmaster.cc, line 191.``` or segmentation fault

### OS Version

MacOS 15.5 (24F74)

royjr, 2025-07-31:
works fine using latest master on macOS 26.0 Beta (25A5316i)

[image]
[image]

josephykwang, 2025-08-01:
is there a way to specify different port other than 9000?