commaai/openpilot #35942: PlotJuggler 2.0

adeebshihadeh, 2025-08-07:
All the features of 1.0 plus
* map
* all cameras
* text logs (from `logcatd`)
* show enums properly
* [radar visualization](https://github.com/commaai/openpilot/blob/master/tools/replay/ui.py)

2.0 doesn't have to necessarily be built on PlotJuggler. Perhaps it makes sense to use another 3rd party tool, build on cabana, or write a new tool entirely.

---

Some nice to haves:
* Lua -> Python
* Rerun and lichblick show the viewer immediately and progressively fill in the data. I think we want this.

adeebshihadeh, 2025-11-28:
@deanlee any interest in this? there's a start from a former intern at `tools/jotpluggler/` that i think should be a good base

deanlee, 2025-11-29:
Oh, I just noticed that Rerun was actually removed in PR #35939.

adeebshihadeh, 2025-11-29:
We found Rerun was both too slow and lacking many simple features (such as search and drag-and-drop). I don't think 

> cameras, radar, 3D views, time sync, logging

Cameras should just be our normal Raylib camera widget. Radar should be a simple port of `tools/replay/ui.py`, and we don't need 3D.