commaai/openpilot #35951: occasional crashes when going onroad

Quantizr, 2025-08-08:
### Describe the bug

launched openpilot with `./launch_openpilot` on a 3X on a jungle on commit `d1e0a60408517750059ac849543fdb1aa7f5b86c` (Aug 5 - a few commits after 0.10 cutoff)

camera never came up and it kicked me off ssh

device restarted itself without pressing anything, but that may have something to do with me launching with `./launch_openpilot` and getting disconnected

there were no issues after booting up again, and I can't reproduce it

several process crashes like this
```
crash
Traceback (most recent call last):
  File "/data/openpilot/openpilot/system/manager/process.py", line 40, in launcher
    mod.main()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 220, in main
    config_realtime_process(4, Priority.CTRL_HIGH)
  File "/data/openpilot/openpilot/common/realtime.py", line 41, in config_realtime_process
    set_core_affinity(c)
  File "/data/openpilot/openpilot/common/realtime.py", line 33, in set_core_affinity
    os.sched_setaffinity(0, cores)
OSError: [Errno 22] Invalid argument
```

no crash log, rlog/qlog was created but both are 0 bytes, boot log is also 0 bytes

attached is the log saved from my terminal before I was disconnected

### Provide a route where the issue occurs

logs were empty

### openpilot version

d1e0a60408517750059ac849543fdb1aa7f5b86c

### Additional info

[ssh_log.txt](https://github.com/user-attachments/files/21676523/ssh_log.txt)

adeebshihadeh, 2025-08-11:
There's not much to go on here.

Do you at least have the bootlog for the subsequent boot?

adeebshihadeh, 2025-08-11:
Subsequent bootlog: `29c92489d9dc0ba9/00000253--ae1bac61dc`

```
[113351.144060] rt_period_timer: expires=113347131139686 now=113347089529213 period=1000000000
[113351.144060] potential CPU hogs:
[113351.144060] current selfdrive.model (337621) is running for 4520208 nsec
[113351.144060] 	selfdrive.model (337621)
[113351.144060] 	selfdrive.model (337714)
[113351.144060] 	selfdrive.model (337723)
[113351.144060] 	selfdrive.model (337724)
[113351.144083] Kernel BUG at ffffff81197046a8 [verbose debug info unavailable]
[113351.144088] ------------[ cut here ]------------
[113351.144091] Kernel BUG at ffffff81197046a8 [verbose debug info unavailable]
[113351.144095] Internal error: Oops - BUG: 0 [#1] PREEMPT SMP
[113351.144100] Modules linked in:
[113351.144109] CPU: 7 PID: 337621 Comm: selfdrive.model Tainted: G        W       4.9.103 #32
[113351.144112] Hardware name: comma tizi (DT)
[113351.144116] task: ffffffe2c8bf4600 task.stack: ffffffe2fe0d0000
[113351.144126] PC is at dump_throttled_rt_tasks+0x1a0/0x1e8
[113351.144129] LR is at dump_throttled_rt_tasks+0x1a0/0x1e8
[113351.144132] pc : [<ffffff81197046a8>] lr : [<ffffff81197046a8>] pstate: 804001c5
[113351.144135] sp : ffffffe2fe0d39c0
[113351.144138] x29: ffffffe2fe0d39c0 x28: ffffffe2c8bf4600 
[113351.144145] x27: ffffff811abc2000 x26: ffffff811b132a98 
[113351.144151] x25: ffffffe33dce4a68 x24: ffffffe33dce4a58 
[113351.144157] x23: 0000000000000064 x22: ffffffe2fe0d3c08 
[113351.144162] x21: ffffffe33dce5048 x20: ffffffe2fe0d3b5e 
[113351.144168] x19: ffffffe33dce5048 x18: 0000000000000010 
[113351.144174] x17: 0000000000a20900 x16: ffffff8119740580 
[113351.144180] x15: ffffffffffffffff x14: 3432373733332820 
[113351.144185] x13: 6c65646f6d2e6576 x12: 697264666c657309 
[113351.144191] x11: 000000000000618e x10: 2e3135333331315b 
[113351.144196] x9 : 0a29333237373333 x8 : ffffff8119a42310 
[113351.144202] x7 : ffffff811c13a8a0 x6 : 00000000000001ca 
[113351.144207] x5 : 0000000000000000 x4 : 0000000000000000 
[113351.144213] x3 : 0000000000000000 x2 : 957230ac3f5c0900 
[113351.144218] x1 : 0000000000000000 x0 : 0000000000000149 
[113351.144224] 
[113351.144224] PC: 0xffffff8119704668:
[113351.144228] 4668  f9400273 eb1302bf 54fffe61 d503201f 110006e2 d2800c81 aa1803e0 93407c42
[113351.144246] 4688  940fc358 71018c1f 2a0003f7 54fffc2d 910153e1 d000d160 91292000 94034199
[113351.144264] 46a8  d4210000 dac002f7 dac012f7 17ffffd4 f9435b00 d5384103 b944f864 cb1402a1
[113351.144281] 46c8  f941f465 d000d162 f9445806 9129a042 aa1403e0 911b0063 cb0500c5 940f9eb7
[113351.144299] 
[113351.144299] LR: 0xffffff8119704668:
[113351.144302] 4668  f9400273 eb1302bf 54fffe61 d503201f 110006e2 d2800c81 aa1803e0 93407c42
[113351.144319] 4688  940fc358 71018c1f 2a0003f7 54fffc2d 910153e1 d000d160 91292000 94034199
[113351.144336] 46a8  d4210000 dac002f7 dac012f7 17ffffd4 f9435b00 d5384103 b944f864 cb1402a1
[113351.144353] 46c8  f941f465 d000d162 f9445806 9129a042 aa1403e0 911b0063 cb0500c5 940f9eb7
[113351.144371] 
[113351.144371] SP: 0xffffffe2fe0d3980:
[113351.144374] 3980  197046a8 ffffff81 fe0d39c0 ffffffe2 197046a8 ffffff81 804001c5 00000000
[113351.144391] 39a0  00000024 00000000 40000000 00000010 ffffffff 0000007f 00000005 00000000
[113351.144408] 39c0  fe0d3c10 ffffffe2 197061c0 ffffff81 c8bf4800 ffffffe2 3dce4a58 ffffffe3
[113351.144425] 39e0  0044f910 00000000 3dce4900 ffffffe3 1abc98f8 ffffff81 1c3bc7b8 ffffff81
[113351.144444] Process selfdrive.model (pid: 337621, stack limit = 0xffffffe2fe0d0000)
[113351.144447] Call trace:
[113351.144451] Exception stack(0xffffffe2fe0d37c0 to 0xffffffe2fe0d38f0)
[113351.144456] 37c0: ffffffe33dce5048 0000007fffffffff ffffffe2fe0d39c0 ffffff81197046a8
[113351.144460] 37e0: 00000000804001c5 000000000000003d ffffffe33dce4a68 0000000119721d30
[113351.144464] 3800: 0000000000000000 ffffff811b134480 ffffffe2fe0d38b0 ffffff8119721f98
[113351.144468] 3820: ffffffe2fe0d3910 ffffff811b132a48 ffffffe33dce5048 ffffffe2fe0d3c08
[113351.144472] 3840: 0000000000000064 ffffffe33dce4a58 ffffffe33dce4a68 ffffff811b132a98
[113351.144475] 3860: ffffff811abc2000 ffffffe2c8bf4600 ffffffe2fe0d3c08 ffffff811c3bdb78
[113351.144479] 3880: 0000000000000000 957230ac3f5c0900 0000000000000149 0000000000000000
[113351.144482] 38a0: 957230ac3f5c0900 0000000000000000 0000000000000000 0000000000000000
[113351.144486] 38c0: 00000000000001ca ffffff811c13a8a0 ffffff8119a42310 0a29333237373333
[113351.144489] 38e0: 2e3135333331315b 000000000000618e
[113351.144494] [<ffffff81197046a8>] dump_throttled_rt_tasks+0x1a0/0x1e8
[113351.144498] [<ffffff81197061c0>] task_tick_rt+0x0/0x138
[113351.144501] [<ffffff8119706504>] dequeue_task_rt+0x2c/0xe0
[113351.144507] [<ffffff81196e465c>] deactivate_task+0x7c/0x190
[113351.144514] [<ffffff811abab828>] __schedule+0x290/0x728
[113351.144518] [<ffffff811ababf78>] schedule+0x38/0x98
[113351.144522] [<ffffff811abaf924>] do_nanosleep+0x94/0x130
[113351.144527] [<ffffff811973dcd0>] hrtimer_nanosleep+0xc8/0x1a8
[113351.144531] [<ffffff811973ef18>] common_nsleep+0x40/0x50
[113351.144535] [<ffffff81197406b8>] SyS_clock_nanosleep+0x138/0x150
[113351.144541] [<ffffff8119683f80>] el0_svc_naked+0x34/0x38
[113351.144545] Code: 910153e1 d000d160 91292000 94034199 (d4210000) 
[113351.144550] ---[ end trace 54f9dfb36420a0d2 ]---
[113351.152976] voc_end_voice_call: Error: End voice called in state 0
```

adeebshihadeh, 2025-08-15:
Removing from the milestone.

This is not reflected in the metrics, and I suspect there was some leftover state while developing the device. openpilot should be more robust and handle that, but there's no evidence this happens in normal use.