commaai/openpilot #35973: `release3-staging` screen blank but car engaged and openpilot is driving the vehicle

yuzisee, 2025-08-11:
### Describe the bug

The device is unresponsive to touchscreen, but other controls such as:

- set speed
- engage/disengage
- etc.

still work fine.

Here is the route in question: 21015c36062c3cef/00000009--edbfacc0d2/22

### Provide a route where the issue occurs

21015c36062c3cef/00000009--edbfacc0d2/22

### openpilot version

release3-staging as of Aug 9

### Additional info

I eventually fixed the issue by stopping the car, unplugging the Comma 3, waiting 30 seconds, and then plugging it back in. It booted up and everything was normal again.

adeebshihadeh, 2025-08-11:
A few questions:
* did it go blank mid-drive or for the whole drive?
* did you ever see the screen on prior, during, or immediately after this drive?
* did you try going onroad/offroad before unplugging the device?

yuzisee, 2025-08-11:
> did it go blank mid-drive or for the whole drive?

Whole drive, iirc

> did you ever see the screen on prior, during, or immediately after this drive?

I saw the screen on previous drives, parked my car in a parking lot (screen still visible), left the car in the parking spot and came back ~2hrs later (usually the screen turns off at some point so nothing seemed out of the ordinary), started the car and started driving (and that's when I noticed the screen was not on -- but I was already on the road so I couldn't really pull over easily to investigate)

> did you try going onroad/offroad before unplugging the device?

I tried stopping the car and turning off the ignition (is that "offroad"?)

This had no effect on the screen.

I resorted to unplugging the device only because turning ignition off/on didn't fix it.

adeebshihadeh, 2025-08-11:
So this route was a fresh boot with bootlog `21015c36062c3cef/00000006--abac956830`. At this point, the best theory is you had a failed display init on boot. This is not a known issue, but I wouldn't be surprised if this still happened super rarely.

I'm going to remove from the milestone as this doesn't seem like a regression at this point.

adeebshihadeh, 2025-12-16:
Closing for now as this is for a comma three, and we haven't heard of any other cases like this.

closed 2025-12-16 by adeebshihadeh