commaai/openpilot #36038: Unable to start the UI following official docs

jasseeeem, 2025-08-21:
### Describe the bug

Hi team. I am new to contributing to Openpilot and was setting it up for the first time following this [official tutorial](https://docs.comma.ai/how-to/turn-the-speed-blue/). But:

- When I run `selfdrive/ui/ui` after running `tools/replay/replay --demo`, I am getting:
```
Error, failed to bind PubSocket to bookmarkButton: Address already in use
Assertion failed: (socket), function PubMaster, file socketmaster.cc, line 191.
[1]    66430 abort      selfdrive/ui/ui
```

- And when I reverse the order and run `tools/replay/replay --demo` after running `selfdrive/ui/ui`, I am getting:

```
active services: accelerometer, accelerometer2, alertDebug, androidLog, audioFeedback, bookmarkButton, cameraOdometry, can, carControl, carOutput, carParams, carState, clocks, controlsState, customReservedRawData0, customReservedRawData1, customReservedRawData2, deviceState, driverAssistance, driverCameraState, driverEncodeData, driverEncodeIdx, driverMonitoringState, driverStateV2, drivingModelData, errorLogMessage, gnssMeasurements, gpsLocation, gpsLocationExternal, gpsNMEA, gyroscope, gyroscope2, lightSensor, liveCalibration, liveDelay, liveParameters, livePose, liveTorqueParameters, liveTracks, livestreamDriverEncodeData, livestreamDriverEncodeIdx, livestreamRoadEncodeData, livestreamRoadEncodeIdx, livestreamWideRoadEncodeData, livestreamWideRoadEncodeIdx, logMessage, longitudinalPlan, magnetometer, managerState, modelV2, navInstruction, navRoute, navThumbnail, onroadEvents, pandaStates, peripheralState, procLog, qRoadEncodeData, qRoadEncodeIdx, qcomGnss, radarState, rawAudioData, roadCameraState, roadEncodeData, roadEncodeIdx, selfdriveState, sendcan, soundPressure, temperatureSensor, temperatureSensor2, testJoystick, thumbnail, touch, ubloxGnss, ubloxRaw, uploaderState, wideRoadCameraState, wideRoadEncodeData, wideRoadEncodeIdx
Error, failed to bind PubSocket to bookmarkButton: Address already in use
Assertion failed: (socket), function PubMaster, file socketmaster.cc, line 191.
[1]    9047 abort      tools/replay/replay --demo
```

I am unable to run both the programs simultaneously.

Similar issue: https://github.com/commaai/openpilot/issues/35855

### OS Version

Mac OS 15.6 (24G84)

### openpilot version or commit

cd9ec6b240abd6e2a2aa4afc29840180a332147f

darkswordz, 2025-08-22:
I am experiencing the same issue. A temporary workaround is to comment  socketmaster.cc:191    //assert(socket);  then rebuild with `scons -j$(nproc)`

jasseeeem, 2025-08-31:
Thanks @jaumebalust! This works fine

paper5590, 2025-11-11:
Shouldn't you have left this issue open? The issue is still present following official docs

Edit: Anyway, I've created a more up-to-date issue report https://github.com/commaai/openpilot/issues/36610 (since they've moved to raylib)