commaai/openpilot #36088: Speaker doesn't work, uninstall/reinstall not enough to fix it, but a reflash fixes it?

nelsonjchen, 2025-09-01:
### Describe the bug

Note that I'm aware of https://flash.comma.ai/. But it's a drastic solution. I am interested in root-cause analysis and would like to know exactly why this issue is happening. Flashing could destroy evidence of why. 

[I mentioned this to @jyoung8607 about a C3X my brother was using in his 2021 CRV Hybrid that had its sound not working.](https://discord.com/channels/469524606043160576/954493346250887168/1410101231610757150) and he suggested I make this issue so we can collect some forensics. I've been dragging my feat.

Today, I did confirm that the sound wasn't working for him when I was driving. When I left, I took his C3X and put it in my 2020 Corolla Hatchback. The sound immediately worked again. I worry I've destroyed some evidence.

It's a bit of a bummer, but I haven't had the chance to move the C3X back to the CR-V to see if the sound stays not working.

It was on `release-staging` and the bug persisted through the 0.10.0 update and was in the version before.

Non-working speaker route on the CR-V: https://connect.comma.ai/fde53c3c109fb4c0/000000a2--5c943de49a

Working speaker route after on my Corolla: https://connect.comma.ai/fde53c3c109fb4c0/000000a4--6c1b015cae

Full logs aren't uploaded yet. I will get that uploaded soon. 🤞 that this issue comes back on the Corolla and/or if we move it back to the CR-V. 

The device did have Sunnypilot on it before switching to release-staging abeit I'm not sure what branch. I'm not sure either how I switched it to release-staging (uninstall/reinstall?). 

Anyway, @jyoung8607 , you got some forensics suggestions, let me know. 

### Provide a route where the issue occurs

https://connect.comma.ai/fde53c3c109fb4c0/000000a2--5c943de49a

### openpilot version

0.10.0 release-staging

MarcoTheDingo, 2025-09-16:
Had the speaker "fail" this morning on my 7 month old c3x, no audio at all. Tried rebooting, issue persists. I'll wait to reflash if you want to grab any diagnostics from my device.

Routes:

- a2152effbad22b70/000002e7--1834c5d0f7
- a2152effbad22b70/000002e8--f6f11f79e3

jyoung8607, 2025-09-16:
The ambient sound level in the car is used to dynamically adjust alert volume. From both devices with the issue, `soundPressure` from `micd` is normal on routes before the issue and flatlined at zero on the affected routes. I'm not sure this actually prevents sound playback, but I can imagine it making alerts too quiet to hear.

The transition point from working to broken on @MarcoTheDingo's device is a little earlier than posted.

Last working: `a2152effbad22b70/000002e3--f9dd63b754`
First broken: `a2152effbad22b70/000002e4--08fdf8c37b`

nelsonjchen, 2025-09-16:
> Had the speaker "fail" this morning on my 7 month old c3x, no audio at all. Tried rebooting, issue persists. I'll wait to reflash if you want to grab any diagnostics from my device.
> 
> Routes (on 2025-09-15 nightly build):
> 
> * a2152effbad22b70/000002e7--1834c5d0f7
> * a2152effbad22b70/000002e8--f6f11f79e3

@MarcoTheDingo Does it persist after a full power down and a power up?

jyoung8607, 2025-09-16:
@nelsonjchen @MarcoTheDingo Have you plugged any USB devices into the 3X aux port? Something along these lines has been triggered in the past by plugging in a Playstation controller for joystick mode, because it has its own microphone. We should be more robust in the face of events like that, but I've yet to understand why your devices are doing it.

nelsonjchen, 2025-09-16:
On the CR-V, there was nothing plugged in.

nelsonjchen, 2025-09-16:
FWIW, it was an outdoor only C3X though. No removing and putting it in the glovebox action FWIW.

jyoung8607, 2025-09-16:
> [@jyoung8607](https://github.com/jyoung8607) , got something for [@MarcoTheDingo](https://github.com/MarcoTheDingo) to run?

While offroad (ignition off), run these commands and share the links that come back. I'm sending the main output off to termbin so you don't have to copy and paste quite so much output.

```
python -c "import sounddevice as sd; print(sd.query_devices())" | nc termbin.com 9999
arecord --list-devices | nc termbin.com 9999
arecord --list-pcms | nc termbin.com 9999
aplay --list-devices | nc termbin.com 9999
aplay --list-pcms | nc termbin.com 9999
```

MarcoTheDingo, 2025-09-16:
Looks like the 8+ hr power discharge fixed it (or the update to today's nightly. Here's the outputs regardless, not sure if they're any help now that it's working.

`comma@comma-42fb99c5:/data/openpilot$ python -c "import sounddevice as sd; print(sd.query_devices())" | nc termbin.com 9999
https://termbin.com/46cd
comma@comma-42fb99c5:/data/openpilot$ arecord --list-devices | nc termbin.com 9999
https://termbin.com/697r
comma@comma-42fb99c5:/data/openpilot$ arecord --list-pcms | nc termbin.com 9999
https://termbin.com/3qvt
comma@comma-42fb99c5:/data/openpilot$ aplay --list-devices | nc termbin.com 9999
https://termbin.com/74kv
comma@comma-42fb99c5:/data/openpilot$ aplay --list-pcms | nc termbin.com 9999
https://termbin.com/3f6n9
comma@comma-42fb99c5:/data/openpilot$ /usr/comma/sound/tinymix controls | nc termbin.com 9999
https://termbin.com/7w2t
comma@comma-42fb99c5:/data/openpilot$ /usr/comma/sound/tinymix contents | nc termbin.com 9999
https://termbin.com/6lxv`

nelsonjchen, 2025-09-18:
Quick demographic question but @tbadgley would you be able to run those commands if needed?

nelsonjchen, 2025-10-26:
Mirrored to Gist and also off Discord a bit as termbin may not be stable hosting. Thanks @jollycar !

> `python -c "import sounddevice as sd; print(sd.query_devices())" | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/bd319ffd272128f8c11f3b165b75a8aa
> 
> `arecord --list-devices | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/e5b1570facb2e3705963c43102050c05
> 
> `arecord --list-pcms | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/fee1f746ce534b9258d026db33813cdc
> 
> `aplay --list-devices | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/d23029ac2b43d6cb3e3c8986299d38e0
> 
> `aplay --list-pcms | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/caf4ee7fb156de2d1479cc132f66128a
> 
> `/usr/comma/sound/tinymix controls | nc termbin.com 9999`
> https://gist.github.com/nelsonjchen/a691abb4d3f2c9a49de907ff652596bc
> 
> `/usr/comma/sound/tinymix contents | nc termbin.com 9999`
> Failed to mixer_ctl_get_array
> Failed to mixer_ctl_get_array
> Failed to mixer_ctl_get_array
> Failed to mixer_ctl_get_array
> ...infinite "Failed to mixer_ctl_get_array" errors...

nelsonjchen, 2025-10-26:
😓 , should have mirrored @MarcoTheDingo 's too, even if it was after the fact.

SunDiego1, 2026-08-18:
Hello, I'm on sunnypilot staging and when I engage it the first time, there is no sound, but it works. If I disengage it and reengage it, it fails to engage and pops this error: micd soundd process not running. I reset a number of times and it will not clear. 

Comma Four
Sunnypilot staging: 2026.003.000

Provide a route where the issue occurs: 28f6181b7a2a9650/00000019--6fb4e82dd7/3

not sure if this is related to this open issue.

nelsonjchen, 2026-08-18:
You should probably report that to the sunnypilot repository.

SunDiego1, 2026-08-18:
Thanks, I did and they said it was probably related to this issue upstream and asked that I post here,. :)

nelsonjchen, 2026-08-18:
Make sure to backlink this issue to the issue on their repo by just mentioning the link.

sunnyhaibin, 2026-08-18:
https://community.sunnypilot.ai/t/micd-soundd-process-not-running-on-staging/6773

dev1837457, 2026-09-26:
We captured the same `sounddevice.PortAudioError: Error querying device -1` from `micd` (and `soundd`) on a comma 4 running AGNOS 19.8 and the September 25, 2026 nightly-dev snapshot. On failing boots, the default audio device was unavailable while the daemons exhausted their startup retries; `sound.target` became active near the end of that window. This supports a startup-ordering race for this case, but does not establish the cause of every silent-audio failure reported here.

We tested a bounded AGNOS launcher wait for `sound.target`, preserved the original daemon retry behavior while logging the underlying exception, and added an optional audio-ready cue. Three parked cold boots passed, followed by a limited drive/engagement check without audio daemon errors or dropouts. The version-specific patch, baseline hashes, test evidence, rollback notes, and limitations are here: https://github.com/dev1837457/openpilot-audio-startup-fix/releases/tag/v0.1.0

This is an independent mitigation, not an upstream fix or a claim that it resolves the persistent speaker failures also discussed in this issue. The repository contains no raw device or route logs.

adeebshihadeh, 2026-11-08:
Should be fixed by https://github.com/commaai/agnos-builder/pull/618. Let us know if it pops up again.

closed 2026-11-08 by adeebshihadeh