commaai/openpilot #36119: Cannot build on macOS

Discountchubbs, 2025-09-08:
### Describe the bug

```scons: *** [tools/cabana/videowidget.o] Error 1
In file included from tools/cabana/tools/routeinfo.cc:6:
In file included from ./tools/cabana/streams/replaystream.h:11:
In file included from ./tools/replay/replay.h:11:
In file included from ./tools/replay/camera.h:10:
In file included from ./tools/replay/framereader.h:9:
./tools/replay/qcom_decoder.h:3:10: fatal error: 'linux/videodev2.h' file not found
    3 | #include <linux/videodev2.h>```

### Provide a route where the issue occurs

N/A (PC)

### openpilot version

8dca43881a9e253158d483dd02a5522563386283

### Additional info

OP master commit: 8dca43881a9e253158d483dd02a5522563386283

Discountchubbs, 2025-09-08:
Mac OS: 15.6.1

jasseeeem, 2025-09-09:
The problem is that `linux/videodev2.h` is not available in MacOS. One solution is to manually add the header file similar to `third_party/linux/include/v4l2-controls.h`, but the problem is: `videodev2.h` has a lot of header files which internally requires even more header files.